flipperdevices/flipperzero-firmware #690: Flipper log

azaitsev, 2021-09-07:
As a user, I'd like to have a log with everything flipper hears in any of reading modes without pressing save button.
I see it as something like:

datetime - app_name - info.to_string()

for example:
2021-01-13 22:45.34 - subghz - Princeton 24bit, key: 0x....
2021-01-13 22:45.38 - lfrfid - em410, hex: 37003...

skotopes, 2021-09-07:
Hi @azaitsev, 

In theory it is possible.
We have logging subsystem which can be modified to keep logs on sd card.

This issue have low priority and we'll be implemented as soon as someone will find free time.

PS: Also PR are welcomed. If someone would want to implement it, I'll explain details.

azaitsev, 2021-09-11:
Sure, I know it's not your top priority yet. I can try to implement it myself as soon as I get my flipper for review.

aprosvetova, 2021-12-23:
Current architecture of `furi-log` subsystem doesn’t allow a simple way to implement this.

We have this idea in a backlog and will notify you if it ever gets implemented. The project is always open for your PRs, so feel free to implement it if you have your own way.

closed 2021-12-23 by aprosvetova