flipperdevices/flipperzero-firmware #706: IDE for development

elektronika-ba, 2021-09-15:
Which IDE is recommended/preferred for flipper firmware development?
Thanks

skotopes, 2021-09-15:
At this point project build and debugging toolchain is console oriented and doesn't depends on IDE.

Here is what we use:
- SublimeText
- Visual Studio Code
- Vim
- Clion
- QtCreator

Any other IDE that is capable to work with external build system is ok.

ErezBinyamin, 2021-10-11:
If someone were to contribute a Flipper0 VScode plugin (or other IDE plugin) what requirements/features would you want to see?

aprosvetova, 2021-12-23:
Someone has set up a Segger Embedded Studio environment, please take a look: https://github.com/hedger/flipperzero-sesproject

We will also think of convenient solutions as we finish working on a new build system which will allow simpler and cross-platform build.

closed 2021-12-23 by aprosvetova