flipperdevices/flipperzero-firmware #848: Snake Game Feedback

OlegSchwann, 2021-11-27:
# What are you want to add or change
https://github.com/flipperdevices/flipperzero-firmware/pull/829 was merged, but since this is the first game, it's not perfect of the box. I have heard some notes about unintuitive timings. As an author who knows the inner workings, I can't see by the user's eyes myself. 

# What questions do you have
Please tell us about your impressions when you first ran the game. What did you expect that didn't come true? Each of us has some kind of experience, where did I not guess it?

OlegSchwann, 2021-12-06:
I asked people: it must not drop button presses. They should be queued up and applied sequentially at a point suitable for turning, not overwritten like they are now.
https://github.com/flipperdevices/flipperzero-firmware/blob/dev/applications/snake-game/snake-game.c#L354-L363

LionZXY, 2021-12-06:
Two main ways to improve snake game:
- Increase snake grid
- Store all pressed button in local storage and repeat it in this order

OlegSchwann, 2021-12-06:
Some discussion here
https://t.me/flipperzero/4248

karasevIA, 2021-12-06:
Difficult, inconvenient, not intuitive, absolutely minimalistic, there is not even a table of records.
And it's perfect!!!
Please don't change anything

aprosvetova, 2021-12-23:
Closing the stale issue.

closed 2021-12-23 by aprosvetova

JulyIghor, 2022-04-24:
Wrong keys behavior. If you press fast, previous direction ignored.
It is need to be a key press queue.
Good example is here https://www.googlesnake.com