flipperdevices/flipperzero-firmware #871: Impossible to upload the firmware when external sound card is connected

okalachev, 2021-12-06:
When a Focusrite Scarlett 2i2 is connected is impossible to upload the firmware. It is inconvenient. I guess these devices may be distinguished somehow.

```bash
$ make -C firmware upload
CFLAGS ok

dfu-util -D .obj/f7/firmware.bin -a 0 -s 0x08008000 
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
dfu-util: More than one DFU capable USB device found! Try `--list' and specify the serial number or disconnect all but one device

make: *** [.obj/f7/upload] Error 74
```

aprosvetova, 2021-12-06:
@zhovner PID conflict. Consider changing PID.

zhovner, 2021-12-07:
@aprosvetova this issue will appear anyway if more than one DFU devices will be attached, not matter what VID/PID it will have.

The only way for you is to specify serial like this:

`dfu-util -D .obj/f7/firmware.bin -a 0 -S -S 206739803753 -s 0x08008000`

where `-S 206739803753` is the serial of Flipper Zero. You can get the serial using this command: `dfu-util --list`

This issue is impossible to fix. qFlipper and Web Update do not have this issue.

closed 2021-12-07 by zhovner

okalachev, 2021-12-07:
@zhovner, it's possible to fix! Check it out: https://github.com/flipperdevices/flipperzero-firmware/pull/877.

PR #877 (2021-12-07): Specify vid and pid of device for upload target
When another DFU device is connected (like a sound card) it's impossible to upload the firmware, which is inconvinient.

This adds VID and PID specification for `dfu-util` to distinguish Flipper from other devices.

Tested with a real Flipper and a Focusrite Scarlett 2i2 connected.

(Fixes #871).