flipperdevices/flipperzero-firmware #874: IRDA doesn't work every time a signal is sent

temamagic, 2021-12-06:
**Describe the bug**

I cloned the signal from my remote and tried to reproduce it. Expected reaction - a signal is sent to each click and the TV reacts, in fact, the TV only responded sometimes, while the remote works well

**To Reproduce**
Steps to reproduce the behavior:
1. Learn new remote
2. Send
3. Wait when tv is on/off

**Expected behavior**
TV reacts every time when you send

**Target**
[sony tv](https://www.sony.ru/electronics/support/televisions-projectors-lcd-tvs/kdl-40ex521)

**Additional context*
Power SIRC A:01 C:15
[image]

[Video bug demo](https://user-images.githubusercontent.com/15264697/144893011-711c84b0-a188-4d7c-af20-ea34f0297088.mp4)

temamagic, 2021-12-06:
Looks like send doesnt work on every time only when you in "learn" menu, saved remotes work fine

temamagic, 2021-12-14:
Yeah, it seems to be true

I checked it manually, it works if you send a lot of signals quickly (lol but, it turns out in the video I did it too)

skotopes, 2022-02-19:
fixed in 0.49.0

Please reopen issue if problem persist.

closed 2022-02-19 by skotopes