flipperdevices/flipperzero-firmware #893: Infrared remote falls on exit

dunaevai135, 2021-12-13:
**Describe the bug**
Если натыкать много раз в кнопку на пульте, он долго это отсылает, и если в этот момент нажать назад, он зависает.

**To Reproduce**
Steps to reproduce the behavior:
1. Открыть remote с кнопками
2. Много раз нажать на ok
3. Пока он отсылает (видно по зелёному морганию и странному звуку от вибромотора ) нажать назад
4. Фриз

**Expected behavior**
Выход на предыдущий уровень меню

**Logs**
```
(gdb) bt
#0  HardFault_Handler () at targets/f7/furi-hal/furi-hal-interrupt.c:188
#1  <signal handler called>
#2  0x00000000 in ?? ()
#3  0x08057a60 in irda_get_new_signal (instance=instance@entry=0x20014498) at /project/lib/irda/worker/irda_worker.c:364
#4  0x08057be8 in irda_worker_tx_thread (thread_context=0x20014498) at /project/lib/irda/worker/irda_worker.c:451
#5  0x0800b876 in furi_thread_body (context=0x20014d08) at /project/core/furi/thread.c:45
#6  0x0806e340 in osMessageQueueDelete (mq_id=0xa5a5a5a5) at ../lib/STM32CubeWB/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2/cmsis_os2.c:1941
#7  0xa5a5a5a4 in ?? ()
```
Похож на разыменование нулевого указателя, из за гонки.

**Target**
dev 195f422bb9c2a1154b511536448d3104202e0e51

albkharisov, 2021-12-13:
Thanks for reporting.

Bug occurs if press short Back before releasing OK button in Saved remotes.
Fixed in https://github.com/flipperdevices/flipperzero-firmware/pull/894

DrZlo13, 2021-12-13:
Fixed in #894

closed 2021-12-13 by DrZlo13