flipperdevices/flipperzero-firmware #899: Bug while interacting with "Settings" and Android App via bluetooth

HydroSausager, 2021-12-14:
**Describe the bug**
While phone is connected to flipper using android app, some interactions with "settings" can break flipper.

**To Reproduce**
[Steps ](https://t.me/flipperzero/9420)to reproduce the behavior:
1 Connect flipper to android app
2 Start screen streaming in app
3 Navigate flipper to "Settings" -> "LCD and notifications" and exit (you can change setting or not) 
4 Flipper freezes until reboot 

(without screen streaming it just breaks ble connection)

**Target**
tested on F6 
firmware - 58ce9373 953 12-12-2021

Uptrend-john, 2022-03-15:
I don't have that exact problem my settings page is just a screen of Red (Android Device)

skotopes, 2022-03-16:
I'll check it after we merge #1021

skotopes, 2022-03-17:
@gornekich @nminaylov reproducible on IOS and Android.
#1033 improves device behaviour but issue still persist in BLE transport for RPC

skotopes, 2022-04-07:
fixed in 0.52

closed 2022-04-07 by skotopes