flipperdevices/flipperzero-firmware #901: IRDA Universal library doesn't sends immediately after open TV

temamagic, 2021-12-14:
**Describe the bug**
When youre open Universal library -> TV and fast click, it doesn't send signal

**To Reproduce**
Steps to reproduce the behavior:
1. Reboot
2. Open Infrared -> Universal library
3. Open TV's and fast click central button (it doesn't work at first time)
3.1 You can click second time and it will be ok
4. Back again
5. Open TV's and wait ~1 sec and then click (sending started)

**Expected behavior**
Sending starts immediately

**Target**
Firmware 0.40.1
Build 1 dec 2021

**Additional context**
My tv.ir looks like: https://gist.github.com/temamagic/ec11a860e433afb98f8e8fe17327598b

**Video Demo**

https://user-images.githubusercontent.com/15264697/146038606-81f24686-658c-4d76-8f74-2ec1f0dcd307.mp4

albkharisov, 2022-02-16:
This is expected behavior for current version. There are background counting of signals in `tv.ir`, so flipper doesn't respond to any input. TV database can be large so it can take time to read whole file.
In future versions there will be a loading animation, notifying about background operations.

skotopes, 2022-02-24:
Fixed in #1004

closed 2022-02-24 by skotopes