flipperdevices/flipperzero-firmware #905: [Sub-Ghz] Do not show "all unsaved data will be lost" screen after saving

temamagic, 2021-12-15:
**Describe the bug**
When you capture the raw signal and save, you will be shown “all unsaved data will be lost”, but it doesn't matter because you saved it before.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Sub-Ghz
2. Open Read Raw
3. Capture signal and save them
4. Press "back" button

**Expected behavior**
If you have saved the signal, there should be no warning.

**Target**
Firmware 0.40.1
Build 1 dec 2021

[image]

[image]

temamagic, 2021-12-19:
UPD: this screen is shown even when you go to "Saved" and select a previously saved "raw" and then exit SAVED

Skorpionm, 2021-12-19:
already done. https://github.com/flipperdevices/flipperzero-firmware/pull/910

closed 2021-12-23 by Skorpionm