flipperdevices/flipperzero-firmware #924: Cant compile firmware with Docker

r3df0xx, 2021-12-22:
**Describe the bug**
I cant compile firmware with Docker, `docker-compose up -d` runs ok, but after `docker-compose exec dev make` i get next output:

**Logs**
```
$ docker-compose exec dev make
make[1]: Entering directory '/project/bootloader'
diff: .obj/f7/BUILD_FLAGS: No such file or directory CFLAGS has been changed

        CC       bootloader/src/main.c -> .obj/f7/main.o
        CC       lib/STM32CubeWB/Drivers/CMSIS/Device/ST/STM32WBxx/Source/Templa
tes/system_stm32wbxx.c -> .obj/f7/system_stm32wbxx.o
        CC       lib/STM32CubeWB/Drivers/STM32WBxx_HAL_Driver/Src/stm32wbxx_ll_u
tils.c -> .obj/f7/stm32wbxx_ll_utils.o
        CC       lib/STM32CubeWB/Drivers/STM32WBxx_HAL_Driver/Src/stm32wbxx_ll_g
pio.c -> .obj/f7/stm32wbxx_ll_gpio.o
        CC       lib/STM32CubeWB/Drivers/STM32WBxx_HAL_Driver/Src/stm32wbxx_ll_i
2c.c -> .obj/f7/stm32wbxx_ll_i2c.o
        CC       lib/STM32CubeWB/Drivers/STM32WBxx_HAL_Driver/Src/stm32wbxx_ll_s
pi.c -> .obj/f7/stm32wbxx_ll_spi.o
        CC       lib/drivers/lp5562.c -> .obj/f7/lp5562.o
        CC       bootloader/targets/f7/furi-hal/furi-hal-gpio.c -> .obj/f7/furi-
hal-gpio.o
        CC       bootloader/targets/f7/furi-hal/furi-hal-i2c-config.c -> .obj/f7
/furi-hal-i2c-config.o
        CC       bootloader/targets/f7/furi-hal/furi-hal-i2c.c -> .obj/f7/furi-h
al-i2c.o
        CC       bootloader/targets/f7/furi-hal/furi-hal-light.c -> .obj/f7/furi
-hal-light.o
        CC       bootloader/targets/f7/furi-hal/furi-hal-resources.c -> .obj/f7/
furi-hal-resources.o
        CC       bootloader/targets/f7/furi-hal/furi-hal-spi-config.c -> .obj/f7
/furi-hal-spi-config.o
        CC       bootloader/targets/f7/furi-hal/furi-hal-spi.c -> .obj/f7/furi-h
al-spi.o
        CC       bootloader/targets/f7/furi-hal/furi-hal-version.c -> .obj/f7/fu
ri-hal-version.o
        CC       bootloader/targets/f7/furi-hal/furi-hal.c -> .obj/f7/furi-hal.o
        CC       lib/toolbox/version.c -> .obj/f7/version.o
        CC       bootloader/targets/f7/target.c -> .obj/f7/target.o
        CC       lib/u8g2/u8g2_glue.c -> .obj/f7/u8g2_glue.o
        CC       lib/u8g2/u8g2_intersection.c -> .obj/f7/u8g2_intersection.o
        CC       lib/u8g2/u8g2_setup.c -> .obj/f7/u8g2_setup.o
        CC       lib/u8g2/u8g2_d_memory.c -> .obj/f7/u8g2_d_memory.o
        CC       lib/u8g2/u8x8_cad.c -> .obj/f7/u8x8_cad.o
        CC       lib/u8g2/u8x8_byte.c -> .obj/f7/u8x8_byte.o
        CC       lib/u8g2/u8x8_gpio.c -> .obj/f7/u8x8_gpio.o
        CC       lib/u8g2/u8x8_display.c -> .obj/f7/u8x8_display.o
        CC       lib/u8g2/u8x8_setup.c -> .obj/f7/u8x8_setup.o
        CC       lib/u8g2/u8g2_hvline.c -> .obj/f7/u8g2_hvline.o
        CC       lib/u8g2/u8g2_line.c -> .obj/f7/u8g2_line.o
        CC       lib/u8g2/u8g2_ll_hvline.c -> .obj/f7/u8g2_ll_hvline.o
        CC       lib/u8g2/u8g2_circle.c -> .obj/f7/u8g2_circle.o
        CC       lib/u8g2/u8g2_box.c -> .obj/f7/u8g2_box.o
        CC       lib/u8g2/u8g2_buffer.c -> .obj/f7/u8g2_buffer.o
        CC       lib/u8g2/u8g2_font.c -> .obj/f7/u8g2_font.o
        CC       lib/u8g2/u8g2_fonts.c -> .obj/f7/u8g2_fonts.o
        CC       lib/u8g2/u8x8_8x8.c -> .obj/f7/u8x8_8x8.o
        CC       lib/u8g2/u8g2_bitmap.c -> .obj/f7/u8g2_bitmap.o
        ASM      lib/STM32CubeWB/Drivers/CMSIS/Device/ST/STM32WBxx/Source/Templa
tes/gcc/startup_stm32wb55xx_cm4.s -> .obj/f7/startup_stm32wb55xx_cm4.o
        LD       .obj/f7/bootloader.elf
   text    data     bss     dec     hex filename
  15096     176    2632   17904    45f0 .obj/f7/bootloader.elf
        HEX      .obj/f7/bootloader.hex
        BIN      .obj/f7/bootloader.bin
        DFU      .obj/f7/bootloader.dfu
/usr/bin/env: 'python3\r': No such file or directory
make[1]: *** [/project/make/rules.mk:53: .obj/f7/bootloader.dfu] Error 127
make[1]: Leaving directory '/project/bootloader'
make: *** [Makefile:44: bootloader_all] Error 2
```

How can I fix that?

skotopes, 2021-12-23:
Hi,

Please check https://github.com/flipperdevices/flipperzero-firmware/pull/923

r3df0xx, 2021-12-23:
> Hi,
> 
> Please check #923

I've merged changes from https://github.com/flipperdevices/flipperzero-firmware/pull/923, but still getting same error as from the start.

r3df0xx, 2021-12-23:
Issue solved, with new 0.44.0rc version build completes OK.

closed 2021-12-23 by r3df0xx

skotopes, 2021-12-23:
Nice )