flipperdevices/flipperzero-firmware #1013: HardFault with particular bank card scan order

brandocrawford36, 2022-03-10:
**Describe the bug**
When scanning bank cards with emv app in a certain order causes flipper to crash. It also only happens when going from my Amazon prime CC (visa) then scanning my CapitalOne CC (mastercard). Same crash happens when going from Prime CC to another CapitalOne card, both of the CapitalOne cards are their QuickSilver line.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on and navigate to NFC > Run special action > Read bank card
2. Read AmazonPrime CC
3. Choose to run EMV app
4. Scan card again and get EMV results then scan CapitalOne and get crash
5. OR Skip scanning Amazon CC and move onto to CapitalOne, get same crash.

I'll add logs when I figure out how.

skotopes, 2022-03-12:
Can you also provide firmware version

brandocrawford36, 2022-03-14:
FW 0.50.0

skotopes, 2022-04-22:
fixed in 0.56.0

closed 2022-04-22 by skotopes