flipperdevices/flipperzero-firmware #1015: Lefty Mode

B3rry, 2022-03-11:
Hi,

I’d like to suggest a lefty mode, as the hardware design of the device makes it difficult for left handed users to operate the flipper zero. Potentially rotating the display output, and 5 way controller’s input 180 degrees would be very helpful.

Astrrra, 2022-03-11:
I second this

skotopes, 2022-03-12:
Feasible. I'l add it to backlog.

alexwaibel, 2022-03-14:
Not sure how feasible this would be given the resolution and current UI design, but it would be cool to be able to rotate the display orientation arbitrarily instead of having a simple lefty toggle. My thought being that if I just flip the device upside down so the buttons are on the left, the IR blaster is no longer facing forward. As a lefty, I'd probably prefer to use the device in the orientation used by the universal remote app (with the dpad at the bottom).

richardbbrooks, 2022-04-28:
Howdy, left-handed here who thought this might be a good first spelunking. I was able to make a proof of concept with just a few changes and confirmed mostly working on my own Zero. Kudos to maintainers for how easy it was to build and flash. It's highly encouraging towards making future contributions. 
[image]

In ./applications/canvas.c changing `U8G2_R0`: (_note this won't survive apps changing orientation, similar changes are needed wherever `u8g2_SetDisplayRotation` is called too_)
```
    u8g2_Setup_st756x_flipper(&canvas->fb, U8G2_R0, u8x8_hw_spi_stm32, u8g2_gpio_and_delay_stm32);
```
to `U8G2_R2`:
```
    u8g2_Setup_st756x_flipper(&canvas->fb, U8G2_R2, u8x8_hw_spi_stm32, u8g2_gpio_and_delay_stm32);
```
Ref: https://github.com/olikraus/u8g2/wiki/u8g2reference#setdisplayrotation

And within ./firmware/targets/f7/furi_hal_resources.c redefining the `key` items `input_pins` so that U/D/L/R are opposites

These changes reveal a few side effects to consider in a polished solution:

- flipping the screen this way also flips the screen data sent over to desktop qFlipper - I am not sure of an elegant solution; it might be easier flip display output within qFlipper after receiving?
- remapping U/D/L/R keys breaks vertical applications like the Universal Remote - it might be nice to resolve this in a way that doesn't require individual application awareness of screen flipping

commit 780da7d4 (2023-03-09): Upside down / left handed orientation support (#2462)

* Add backup files to .gitignore
* Added lefty support in Settings > System > hand Orient: Fixes: #1015
* Left handed mode
* Fix lefthanded mode on vertical interfaces
* Input: new composite sequence identifier
* Gui: move input mapping from Canvas to ViewPort, properly handle input mapping on View switch in ViewDispatcher
* Rpc: proper input sequencing and tagging in RpcGui
* Rpc: remove magic from RpcGui
closed 2023-03-09 by skotopes