flipperdevices/flipperzero-firmware #1019: Hard crash reading Monzo beta card (NFC) (MasterCard EMV)

hakusaro, 2022-03-13:
**Describe the bug**
I ran into a hard crash when attempting to read my Monzo beta card (via NFC). I have a video of this, but it's as straightforward as these things go: attempting to read the card as a bank card crashes with `furi_check failed`.

This happens on version 0.50.0 [03-03-2022] / 3f164ef3 [1103], 100% of the time with this card.

I turned on debugging and debug logs set to info level, but I didn't see any logs generated in any obvious places on the SD card or anywhere that I could find accessible in the menus itself. If you give me instructions, I would be more than happy to provide more logs.

Also, out of the 5 or so cards I've tried, this is the only card that hard crashes like this. I haven't had success running the EMV apps from all cards, and it seems like MasterCard is the most unreliable. Visa seems fine.

Video: https://streamable.com/lsc4dk

jman1118, 2022-03-15:
I have the same issue when reading cards from Apple Pay.

hakusaro, 2022-03-15:
@jman1118 I don’t actually have this problem reading Apple Pay cards from Visa. I haven’t tried MasterCard though. What’s the issuer?

jman1118, 2022-03-15:
Any of my physical mastercards didnt read on the first try. Earlier apple pay locked up the flipper but havent been able to reproduce this.

jman1118, 2022-03-15:
My apple card doesnt read at all but that could be because its metal.

gteague, 2022-03-15:
apple card i get a popup with nfc-a as the first line and then it refuses to go any further. /guy

hakusaro, 2022-03-15:
The physical Apple Card doesn't have an NFC chip inside it, iirc.

gteague, 2022-03-15:
ah! i was trying about a half-dozen cards and only the visa worked as i expected. afterwards i tried reading an air-tag just for s&g's and that was the one which only responded to the /read/ option and said it was a bank card, but wouldn't go any further. and the /bank card/ read option didn't work. sorry but my short term memory isn't what it used to be. and i know zero about this stuff to begin with. /guy

hakusaro, 2022-03-24:
Fwiw, this still happens on `0.51.0` / f2a84851.

gornekich, 2022-04-20:
Unfortunately we can't reproduce the issue. Meanwhile I made NFC app refactoring, now it is in dev branch and soon it will be in release. Could you reproduce the problem with updated firmware? If the problem is not solved it will be very helpful if you provide console logs with Debug level

hakusaro, 2022-04-22:
@gornekich it still crashes on 0.56.0 RC, and I still have no idea how to capture logs after enabling them.

hakusaro, 2022-04-24:
Still crashes on `0.56.1` RC!

hire-jasonrush, 2022-05-04:
I've got 3x bank cards that show NFC. One reads, one is ignored, and one crashes with the same error @hakusaro originally reported here. I've just upgraded to 0.57.0 and can confirm that same card still crashes the bank card reading "app".

makeworld-the-better-one, 2022-05-06:
I am also experiencing this with a Mastercard, using 0.57.0. Here is my log output, with Log Level: Debug and Debug: Enable.

```
59869 [I][LoaderSrv]: Starting: NFC
59878 [I][AnimationManager]: Unload animation 'L1_Recording_128x51'
60762 [D][DolphinState]: icounter 54, butthurt 0
65354 [D][DolphinState]: icounter 54, butthurt 0
67917 [D][DolphinState]: icounter 54, butthurt 0
68927 [D][NfcWorker]: Can't find any cards
69951 [D][NfcWorker]: Can't find any cards
70145 [D][Emv]: Send select PPSE
70216 [D][DolphinState]: icounter 54, butthurt 0
72961 [D][DolphinState]: icounter 54, butthurt 0
73144 [D][Emv]: Send select PPSE
73180 [D][Emv]: Start application
73215 [D][Emv]: Get proccessing options
73272 [D][Emv]: Search PAN in SFI
73643 [D][Emv]: Send select PPSE
73676 [D][Emv]: Start application
73712 [D][Emv]: Get proccessing options
```

Flipper then crashes, and when loading back up, the screen says `HardFault`.

I also have the log with Log Level: Trace. But it seems that would contain confidential data like card number. Let me know if the Flipper team wants that log, and what lines I can remove to anonymize the data.

hakusaro, 2022-05-06:
@makeworld-the-better-one where did you get the log data from? I'd be more than happy to share mine if I can find it. I've tried enabling debug log options, but I haven't found any files on the internal memory or SD card with logs, and connected to qFlipper, I don't see any log output there. o.o

makeworld-the-better-one, 2022-05-06:
@hakusaro I think you should see it in qFlipper. But personally I plugged mine in and connected to the serial port, then ran the `log` command. On Linux I ran `picocom -b 115200 /dev/ttyACM0`, on Windows you could use the PuTTY software. If you are having issues or need help I'd recommend asking on the Flipper Zero Discord.

hire-jasonrush, 2022-05-07:
[qFlipper-20220506-162831.txt](https://github.com/flipperdevices/flipperzero-firmware/files/8644171/qFlipper-20220506-162831.txt)

After finding how to enable trace logs, I've got more detailed info as well. When I run 'log' from the CLI and then run the "Read Bank Card" action, I get the following:

958762 [T][FuriHalNfc]: Current state 20
958763 [T][FuriHalNfc]: Timeout
958765 [D][NfcWorker]: Can't find any cards
958789 [T][FuriHalNfc]: Current state 10
... x43 duplicate lines removed
958929 [T][FuriHalNfc]: Current state 10
958931 [T][FuriHalNfc]: Current state 11
... x6 duplicate lines removed
958960 [T][FuriHalNfc]: Current state 11
958962 [T][FuriHalNfc]: Current state 13
... x4 duplicate lines removed
958975 [T][FuriHalNfc]: Current state 13
958978 [D][Emv]: Send select PPSE
959008 [T][Emv]: Select PPSE answer:
... A bunch of hex I've tentatively left out in case it has card number/etc information.
959044 [D][DolphinState]: icounter 54, butthurt 0
959047 [T][StorageAPI]: File/Dir 0x10ed0 alloc
959054 [T][StorageAPI]: File 0x10ed0 - 0x11b8c open (/ext/nfc/assets/aid.nfc)
959220 [T][StorageAPI]: File 0x10ed0 - 0x11b8c closed
959223 [T][StorageAPI]: File/Dir 0x10ed0 free
... At this point, the serial connection dropped.

hire-jasonrush, 2022-05-07:
If you're running Windows, I've also created instructions (no screenshots yet) on how to gather various logs here: https://flipperzero.miraheze.org/wiki/How_to_gather_logs#Flipper_logs

SnowLeopard71, 2022-05-09:
I also have a card that causes a **HardFault**, BIN 533866.  (FYI, [BIN](https://www.bincodes.com/) ) The debug output is similar to [above](https://github.com/flipperdevices/flipperzero-firmware/issues/1019#issuecomment-1119843193).  (I've set logging to *trace* for maximum verbosity.)
A working card only goes through the *do-while* in *emv_read_bank_card()* (lib/nfc_protocols/emv.c) once, but this one does it twice and the second time, the response *Get processing options answer:* is empty.
The other significant difference in the output of working vs crashing card is the length of the *Start application answer* is 23 bytes shorter for the crashing card.

SnowLeopard71, 2022-05-10:
Found a fix.  Incrementing the index by 2 at line 295 in _emv_decode_read_sfi_record()_ for the country_code makes it skip over the CC number -- there is one "padding" byte less after the country_code on the crashing cards compared to non-crashing ones.  
I changed to increment by 1 and compiled a test and all my cards work fine now.  (I'd submit a pull request but it's been so long I don't remember all the steps properly.)
[fixed firmware](https://x1z.net/flipper-z-f7-full-local-f04d0eea-dirty.dfu) in case anyone wants to test.

SnowLeopard71, 2022-05-12:
The new release, 0.57.1, should fix the `HardFault` crash, but the` furi_check failed` probably still exists.
I also have a CapitalOne MC (545756) where the EMV app loops forever.  The response to _Start application_ is different and does not even include the card number like other working cards.

SnowLeopard71, 2022-05-12:
Here.     To avoid confusion, this is debug output for BIN 545756 that loops forever but no crash.
I've modified the _emv_trace()_ to also output what is sent (TX) as well as received (RX).  The application answer has the PDOL (9F 38) but _emv_decode_select_app_response()_ is missing it -- also added extra logging in the _else if_ to log when it gets the PDOL which is not in this output but is with good cards).
```
56389 [D][Emv]: Send select PPSE
56423 [T][Emv]: Select PPSE answer:
TX: 00 A4 04 00 0E 32 50 41 59 2E 53 59 53 2E 44 44 46 30 31 00
RX: 6F 2F 84 0E 32 50 41 59 2E 53 59 53 2E 44 44 46 30 31 A5 1D BF 0C 1A 61 18 4F 07 A0 00 00 00 04 10 10 50 0A 4D 41 53 54 45 52 43 41 52 44 87 01 01 90 00
56428 [D][Emv]: Start application
56460 [T][Emv]: Start application answer:
TX: 00 A4 04 00 07 A0 00 00 00 04 10 10 00
RX: 6F 4B 84 07 A0 00 00 00 04 10 10 A5 40 50 0A 4D 41 53 54 45 52 43 41 52 44 87 01 01 9F 11 01 01 9F 12 0B 43 41 50 49 54 41 4C 20 4F 4E 45 9F 38 03 9F 40 05 5F 2D 04 65 6E 66 72 BF 0C 0F 9F 4D 02 0B 0A 9F 6E 07 01 24 00 00 30 30 00 90 00
56465 [D][Emv]: Get proccessing options
56503 [T][Emv]: Get processing options answer:
TX: 80 A8 00 00 02 83 00 00
RX: 67 00
56507 [E][Emv]: Failed to decode processing options

```

skotopes, 2022-05-12:
@gornekich could you please check

hakusaro, 2022-05-12:
Okay, I've reproduced my failure on 0.58.1, this time with logs (on macOS, with `picocom -b 115200 /dev/tty.usbmodemflip_[name]`):

```
Press CTRL+C to stop...
212434 [I][LoaderSrv]: Starting: NFC
212443 [I][AnimationManager]: Unload animation 'L1_Read_books_128x64'
217497 [D][DolphinState]: icounter 233, butthurt 3
217504 [T][FuriHalNfc]: Current state 10
217515 [T][FuriHalNfc]: Current state 10
217520 [T][FuriHalNfc]: Current state 10
217523 [T][FuriHalNfc]: Current state 10
217527 [T][FuriHalNfc]: Current state 10
217531 [T][FuriHalNfc]: Current state 10
217535 [T][FuriHalNfc]: Current state 10
217539 [T][FuriHalNfc]: Current state 10
217542 [T][FuriHalNfc]: Current state 10
217544 [T][FuriHalNfc]: Current state 10
217550 [T][FuriHalNfc]: Current state 10
217557 [T][FuriHalNfc]: Current state 10
217560 [T][FuriHalNfc]: Current state 10
217562 [T][FuriHalNfc]: Current state 10
217570 [T][FuriHalNfc]: Current state 10
217575 [T][FuriHalNfc]: Current state 10
217578 [T][FuriHalNfc]: Current state 10
217580 [T][FuriHalNfc]: Current state 10
217585 [T][FuriHalNfc]: Current state 10
217590 [T][FuriHalNfc]: Current state 10
217592 [T][FuriHalNfc]: Current state 10
217594 [T][FuriHalNfc]: Current state 10
217596 [T][FuriHalNfc]: Current state 10
217598 [T][FuriHalNfc]: Current state 10
217600 [T][FuriHalNfc]: Current state 10
217602 [T][FuriHalNfc]: Current state 10
217605 [T][FuriHalNfc]: Current state 10
217607 [T][FuriHalNfc]: Current state 10
217609 [T][FuriHalNfc]: Current state 10
217621 [T][FuriHalNfc]: Current state 10
217626 [T][FuriHalNfc]: Current state 10
217628 [T][FuriHalNfc]: Current state 10
217630 [T][FuriHalNfc]: Current state 10
217636 [T][FuriHalNfc]: Current state 10
217641 [T][FuriHalNfc]: Current state 10
217644 [T][FuriHalNfc]: Current state 10
217646 [T][FuriHalNfc]: Current state 10
217651 [T][FuriHalNfc]: Current state 10
217653 [T][FuriHalNfc]: Current state 11
217661 [T][FuriHalNfc]: Current state 11
217664 [T][FuriHalNfc]: Current state 11
217671 [T][FuriHalNfc]: Current state 11
217684 [T][FuriHalNfc]: Current state 11
217693 [T][FuriHalNfc]: Current state 11
217706 [T][FuriHalNfc]: Current state 11
217711 [T][FuriHalNfc]: Current state 11
217714 [T][FuriHalNfc]: Current state 11
217717 [T][FuriHalNfc]: Current state 13
217722 [T][FuriHalNfc]: Current state 13
217724 [T][FuriHalNfc]: Current state 13
217727 [T][FuriHalNfc]: Current state 13
217729 [T][FuriHalNfc]: Current state 13
217731 [T][FuriHalNfc]: Current state 13
217733 [D][Emv]: Send select PPSE
217765 [T][Emv]: Select PPSE answer:
[redacted -- if this isn't sensitive let me know]
217801 [D][DolphinState]: icounter 236, butthurt 3
217805 [T][StorageAPI]: File/Dir 0x11390 alloc
217813 [T][StorageAPI]: File 0x11390 - 0x114d4 open (/ext/nfc/assets/aid.nfc)
218010 [T][StorageAPI]: File 0x11390 - 0x114d4 closed
218012 [T][StorageAPI]: File/Dir 0x11390 free

FATAL: read from port failed: Device not configured
term_exitfunc: reset failed for dev UNKNOWN: Device not configured
```

I think we've got a couple disparate failures in different sections here.

makeworld-the-better-one, 2022-05-13:
> The new release, ~~0.57.1~~ 0.58.1, should fix the `HardFault` crash  
> \- @SnowLeopard71 

Can confirm, the MasterCard I was scanning no longer causes a crash with the new firmware. Thanks for the PR!

skotopes, 2022-07-10:
Please reopen if issue is still there

closed 2022-07-10 by skotopes

hakusaro, 2022-07-10:
Oops! Thank you! It’s working on my test card too now!