flipperdevices/flipperzero-firmware #1020: 3v3 and 5v GPIO Output Issue

MuddledBox, 2022-03-14:
Playing around with the GPIO today I’ve noticed a few oddities, either the GPIO commands via serial are non-functional or not working as expected, and providing odd functionality.

---

Scenario Name: Turn off 3v3 voltage via Serial
Flipper Zero Firmware: 0.50.0 0.50.0 (3f164ef3 built on 03-03-2022)
Test Setup: Multi-meter plugged into Pin 11 (GND) and Pin 9 (3v3), Flipper connected via USB-C
Additional Software Used: PuTTY
Repro Steps:

Plug in multi-meter to Pin 11 and Pin 9. Connect Flipper to computer and turn on. Open PuTTY and connect to relevant COM port. Type “power 3v3 1”. Observe voltage on multi-meter as 3.30v, as expected.

Next, type “power 3v3 0”. Observe voltage on multi-meter as 2.81v, not as expected. Expected result was 0.00v reading on multi-meter.

---

Scenario Name: Turn off 5v voltage via Serial
Flipper Zero Firmware: 0.50.0 0.50.0 (3f164ef3 built on 03-03-2022)
Test Setup: Multi-meter plugged into Pin 11 (GND) and Pin 1 (5v), Flipper connected via USB-C
Additional Software Used: PuTTY
Repro Steps:

Plug in multi-meter to Pin 11 and Pin 1. Connect Flipper to computer and turn on. Open PuTTY and connect to relevant COM port. Type “power 5v 1”. Observe voltage on multi-meter as 4.96v-5.04v, as expected (within variance).

Next, type “power 5v 0”. Observe voltage on multi-meter as 4.96v-5.04v, not as expected. Expected result was 0.00v reading on multi-meter.

---

In summary, clearly something wrong here. For sanity, I also tested via the “5V on GPIO” functionality under GPIO and again, GPIO voltage was always on, never went to 0.00v as expected. I also noted some variance from the 3v3 pin where voltage reached up to 5.26v briefly, but I’m not sure why that would be or how to recreate.

skotopes, 2022-03-14:
3.3V: this line is used to power SD-Card it is highly unrecommended to turn this line off(power will be re-enabled to scan sdcard). We will remove this command in future releases.
5V: it is expected behaviour when USB connected. Also using iButton will enable 5V while reading keys.

MuddledBox, 2022-03-14:
Makes sense, thanks!