flipperdevices/flipperzero-firmware #1023: No option to save scanned NFC-V, only Retry button

alexlanghart, 2022-03-14:
**Describe the bug**
On an earlier firmware version I was able to scan this same card and save it (whichever out-of-the-box FW version it was), but after updating to v0.50.0 it doesn't display any button to save it. See screenshot

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `NFC`
2. Go to `Read card`
3. Scan this card (which is a gym membership door card)
4. Note that it reads the card successfully but only presents a `Retry` button - no button to save it.

**Expected behavior**
Upon successful reading of an NFC card, it should present an option to save the card.

**Logs**
Add debug logs

**Target**
Specify the target

**Additional context**
[image]
[image]

skotopes, 2022-03-16:
Looks like it came with NFC refactoring, we will return this option in future releases.
Do you have use cases for UID emulation?

alexlanghart, 2022-03-16:
I suppose a use case would be having a single device for multiple NFC ID cards, or backing them up.

skotopes, 2022-03-16:
But you do understand that what we saving here is only card's UID?
Saving real card data is something that we plan to implement in future.

Also what card are we talking about right now?

alexlanghart, 2022-03-16:
Oh I assumed the UID was the card's full data; it's simply a gym door access card, so I wouldn't be surprised if it's only a UID.

skotopes, 2022-03-16:
Gym most likely just uses UID

berkgercek, 2022-12-05:
gonna dredge this issue back up because I would also like NFC-V emulation support. I've run into a system where I'd really like to emulate my own NFC-V card.

Is there an interim way of just emulating the UID until there is full NFC-V emulation support?

skotopes, 2022-12-05:
We remember about this issue. NFC-V/B/F are in backlog, we'll start working on them after NFC HAL refactoring(currently in progress).

nezza, 2022-12-21:
To adjust expectations, my current understanding is that it's not known whether emulating NFC-V is possible using the current hardware, and if it is it will be pretty complex, correct? (NFC-B & F should be supported by the ST25R3916)

r12f, 2022-12-30:
yea, it looks like only NFC-V reader is supported, but not emulator: https://www.st.com/en/nfc/st25r3916.html.

MadjidZ, 2023-05-05:
It seems to be happening on this branch : https://github.com/g3gg0/flipperzero-firmware/tree/ISO15693

skotopes, 2023-06-22:
And was merged some time ago

closed 2023-06-22 by skotopes