flipperdevices/flipperzero-firmware #1029: Various HID formats: detection

dc540, 2022-03-16:
From my post on Reddit https://www.reddit.com/r/flipperzero/comments/tf312l/fun_new_toy_time_rfid_questions/i0up2vt/?context=3

Some HID formats on LF tags are detected, others are not.

Detected: 26bit HID Prox 
Not detected: 34bit HID Prox

If there's more info you need, I have a proxmark3 RDV4 with latest iceman firmware. I may have additional tags I can test, and/or I may be able to program tags other LF formats if that helps.

DrZlo13, 2022-03-16:
Welcome!
Do you have 34-bit HID fobs? Dumps of this format would help me a lot, I need as many of them as you can take.

zzhang1, 2022-03-16:
Based on another thread on Reddit there was a good comment linking to an article describing 125khz formats with different bit lengths in addition to 26 and 34. As that poster suggests, maybe another interim solution is to try to implement a raw read/playback feature if that's easier than handling all the different formats. https://www.reddit.com/r/flipperzero/comments/tc7eoy/comment/i0p07uv/

DrZlo13, 2022-03-16:
`implement a raw read/playback feature`
I don't think this will be an adequate working solution. LF data is very noisy, and besides, it is a continuous stream of data, and many readers rely on this fact. We do not know where the beginning or where the end is, and we do not even know the length of the data. 
But in any case, it needs to be researched.

DrZlo13, 2022-03-16:
@dc540 i need full log, with raw data. Something like this:

```
[usb] pm3 --> lf search

[=] NOTE: some demods output possible binary
[=] if it finds something that looks like a tag
[=] False Positives ARE possible
[=]
[=] Checking for known tags...
[=]
[+] [H10306] - HID H10306 34-bit;  FC: 4660  CN: 22136    parity: valid
[+] [N10002] - HID N10002 34-bit;  FC: 52  CN: 22136
[+] [Optus34] - Indala Optus 34-bit;  FC: 1656  CN: 4660
[+] [Smartpass] - Cardkey Smartpass 34-bit;  FC: 582  CN: 22136  Issue 4
[+] [BQT] - BQT 34-bit;  FC: 18  CN: 3430008    parity: valid
[=] raw: 00000000000000262468acf1

[+] Valid HID Prox ID found!
```

dc540, 2022-03-16:
Interesting. Your question sent me on an interesting journey.

iceman's firmware does not include the raw data in output of lf search, just the data I provided. So I thought, well, maybe the official firmware does. But when I flashed to the official firmware and ran its corresponding client, lf search reports the following:

Checking for known tags:
          
HID Prox TAG ID: 2401c1778e          
Invalid or unsupported tag length.          

Valid HID Prox ID Found!          

Valid T55xx Chip Found
Try lf t55xx ... commands

Even on the 26-bit card that is recognized better, the official firmware is not reporting anything beyond what I provided using lf search.  No raw data, no specific tag sub-type info, nothing.

Is there a firmware you recommend?

dc540, 2022-03-16:
in any case you should be able to clone a HID tag using that ID and test on that.

DrZlo13, 2022-03-17:
@dc540 
Unfortunately proxmark sometimes ignores important data, so raw data from the card would be very helpful to me.
Please try"
```
lf hid reader
```
```
lf hid demod
```

dc540, 2022-03-17:
Sorry, I'm still not sure I'm using the same proxmark firmware that you're used to. Neither the official firmware nor the iceman fork are reporting what you seem to be expecting.

Official firmware using the above commands (after changing "lf hid reader" to "lf hid read") still reports "unsupported tag length" 

While iceman's latest reports slightly better, it still does not include raw data anywhere in the output:

pm3 --> lf hid read
HID Prox TAG ID: 2401c1778e (48071) - Format Len: 34bit - FC: 224 - Card: 48071          
pm3 --> lf hid demod
HID Prox TAG ID: 2401c1778e (48071) - Format Len: 34bit - FC: 224 - Card: 48071          

If there's a firmware and client that reports more, I'm happy to use it, I just need to be pointed to it. 

I was able to confirm something, though --
the official firmware includes the encode functionality which confirms the tag ID based on the facility code, card ID and suspected format (N10002).

proxmark3> lf hid encode N10002 f 96 c 48071
HID Prox TAG ID: 2400c1778e     

Iceman's fork does not seem to include that functionality.

DrZlo13, 2022-03-17:
I use the firmware from https://github.com/RfidResearchGroup/proxmark3.git, it seems to be the most alive..

drewbeer, 2022-03-22:
here is another one that i just pulled, a valuprox hid card.

`[+] [H10302  ] HID H10302 37-bit huge ID          CN: 11850046624  parity ( ok )
[+] [H10304  ] HID H10304 37-bit                FC: 22602  CN: 89248  parity ( ok )
[+] [P10004  ] HID P10004 37-bit PCSC           FC: 2825  CN: 71114
[+] [MDI37   ] PointGuard MDI 37-bit            FC: 6  CN: 38886560  parity ( ok )
`

anders8, 2022-04-27:
Ooooo, this feels so close! Thanks for the research @dc540 and @drewbeer , and thanks for coding @skotopes !!! ❤ 😎🗝

skotopes, 2022-09-06:
Check latest release.

closed 2022-09-06 by skotopes