flipperdevices/flipperzero-firmware #1031: SubGHz: return 300MHz frequency

wbraswell, 2022-03-17:
Tag @Skorpionm 
Chamberlain brand garage door remote control
FCC ID: hbw2281
300MHz

skotopes, 2022-03-17:
@wbraswell by the way, it's already fixed in #1028

wbraswell, 2022-03-17:
@skotopes 
Without access to Jira, I can't find any way to confirm #1028 actually adds back in 300MHz?

MuddledBox, 2022-03-17:
@wbraswell look in https://github.com/flipperdevices/flipperzero-firmware/pull/1028/files

Specifically: 
[applications/subghz/subghz.c](https://github.com/flipperdevices/flipperzero-firmware/pull/1028/files#diff-7655a7acf96feb4c136f87657511de7dd28d88749c6d7244ccbe22adc786cec9)

skotopes, 2022-03-17:
@wbraswell we currently migrating to github issues, we do not plan to continue to use jira.

closed 2022-03-17 by skotopes