flipperdevices/flipperzero-firmware #1037: [Feature Request] Intigrate haystacks as a background process

fennectech, 2022-03-17:
https://github.com/seemoo-lab/openhaystack

This project emulates air tags   it woud be awesome if the flipper could act as an air tag when bluetooth is on   there are so many ways this couold be useful

Astrrra, 2022-03-17:
This was already done in december in https://github.com/AlexStrNik/flipperzero-firmware, but as far as I remember, using OHS blocked any other bluetooth features, and the whole project became really hard to maintain after some low-level changes to the upstream bluetooth code

fennectech, 2022-03-17:
I would be fine with it being something you could turn on and off.   Would you guys please be wiling to bring this back?    
I imagined something like this 

**Bluetooth Mode**
`Flipper Companion app
Open Haystack
OFF
`

It would eliminate the problem of OHS blocking other features as its an option the user can select and change at will.

Astrrra, 2022-03-17:
That's exactly how it works in the fork, but then again, it's quite hard to maintain

fennectech, 2022-03-17:
is there any chance it will be brought forward into the current firmware

Astrrra, 2022-03-17:
Probably not, switching between those two modes will be a major inconvenience when the companion apps are finished

You're free to implement this in your own fork though! I think looking at the old code (link above) might help a lot with that

skotopes, 2022-03-20:
Hi.

As it was already mentioned it requires different radio firmware which at this point is impossible(not enough free space to use full radio stack).

We will come back to this idea after we add loading apps from SD.

closed 2022-03-20 by skotopes

CookieCr2nk, 2022-10-15:
As now SD-Card Apps are available, maybe it could be possible.