flipperdevices/flipperzero-firmware #1038: Feature Request: i2c and SPI bridge to USB

Patronics, 2022-03-17:
Similarly to the USB-UART bridge, it'd be really useful to have an option to use flipper as a i2c device (Ideally both as a master and a slave that shares the memory to the host computer). Similarly a SPI Bridge option would be useful.

skotopes, 2022-03-20:
Hi. What protocol do you propose?

Patronics, 2022-03-20:
Hmm, I'm not too familiar with the specific options, but perhaps it'd make sense to emulate the FTDI FT232H? It is used for i2c and SPI, in addition to some other protocols like JTAG and UART. If there's another option that's easier to implement that'd be fine too, as long as the functionality of enabling i2c and SPI communication from the computer is available. https://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT232H.pdf

skotopes, 2022-03-20:
@Patronics FTDI is closed source product, it's not very good idea.

Patronics, 2022-03-20:
Hmm, what about this open source project? https://github.com/harbaum/I2C-Tiny-USB

skotopes, 2022-03-20:
Looks like they have their own protocol and using low level access to USB. Theoretically possible.
I think it is a good topic for research, it will be nice to find all available options before implementing it.

Patronics, 2022-04-19:
I found another good option, the "Binho Nova Multi-Protocol USB Host Adapter"  with a pretty friendly and intuitive (though not open source, sadly) implementation, emulating a UART adapter/virtual com port that implements the other commands internally, so there wouldn't need to be any new drivers on the host device, and it already has integration with python libraries). The ascii interface is documented here: https://support.binho.io/user-guide/ascii-interface.

skotopes, 2022-04-19:
Conversion to text and back will be slow.
We will add option to work with i2c and spi throw protobuf and provide library to work with it.

tiagogala, 2022-07-14:
I2C master can (more or less) be implemented in a similar fashion to the i2cget/i2cset/i2cdetect commands on Linux: https://linux.die.net/man/8/i2cdetect
That shouldn’t be terribly hard.
I2C slave probably can be done by defining static a map of register/value. Not sure how that should work for dynamic data.

incyi, 2022-07-14:
https://www.adafruit.com/product/4267

Something like this ? 
Here is the source:

https://github.com/jamesbowman/i2cdriver