flipperdevices/flipperzero-firmware #1039: Feature request: GPIO Logic analyzer/scope

Patronics, 2022-03-17:
A really useful option for debugging circuits, a logic analyzer app (and possibly even an analog scope option) would be very powerful! Could also optionally output the data to a computer for more advanced analysis, perhaps mimicking either a Saleae logic analyzer or a cheaper knock-off one.

skotopes, 2022-03-20:
We've been exploring that area for a while and found that it's not going to be very useful because of hardware limitations(ARM core + USB + Memory performance). But do plan to add some basic scope to GPIO screen.

drakes00, 2022-11-03:
Hi,

@skotopes Were you able to initiate something regarding this feature? I'd be interested to contribute on if not but I never implemented something for flipper zero. My goal to start with would be to add a screen where one or more gpio are displayed in high or low state in real time. 

Could you provide me some guidelines to start on something?

Thanks a lot

skotopes, 2022-11-17:
@gornekich can share some thought

gornekich, 2022-11-28:
Hello @drakes00 !
I was thinking about Logic Analyzer app too. As @skotopes said, it might not be perfect because of hardware limitation. However there is one approach that can be used to capture signal with minimal CPU resources. You can run TIM2 (32bit timer) and trigger DMA through DMAMUX by EXTI signal to transfer timings from timer register to memory. At the end you will get your digital signal representation as an array of timings for level durations.
One of our community member have basic implementation for this approach. I want to ask him to push his code. This would be a good start for writing Logic Analizer App

biotinker, 2022-12-10:
I would also find this functionality very useful.

I've been working on a fork of the stock GPIO app which has GPIO input support: https://github.com/biotinker/flipperzero-gpioreader

Plan is to add analog/scope features to it as well, though if other people are already working on such things, I'd rather team up than duplicate work.

Is there any plan to add ADC functionality to FuriHal in the near term? That would make the job much, much easier.

gabbapeople, 2023-01-12:
Waiting for ADC functionality to FuriHal very much! =)

skotopes, 2024-01-03:
I guess scope/logic analyzer already implemented as apps in catalog. I'll close this issue.

closed 2024-01-03 by skotopes