flipperdevices/flipperzero-firmware #1040: Feature Request: "Twin Duck" usb rubber ducky emulation

Patronics, 2022-03-17:
Many of the sample badusb scripts from Hak5 use the "Twin Duck" firmware, emulating a USB drive in addition to a keyboard input. This would add a lot more flexibility to the BadUSB feature.

MuddledBox, 2022-03-18:
This has been requested a bit in the Discord. 

Currently the SD card in the Flipper Zero is not recognized as a mass storage device by Windows. That would probably be a good step one before implementing this feature.

skotopes, 2022-03-20:
Theoretically possible. But going to be suuuuuuper slow.
We'll keep this thing in backlog, but don't expect it be implemented anytime soon.

Woodie4, 2022-08-20:
+1 This would be amazing... I really hope this will get implemented

sethpeace, 2022-08-26:
DuckyScript 3.0 transfers data by flashing the CapsLock and NumLock LEDs, meaning the OS wouldn't have to use the card as mass storage at all. Maybe you could implement it that way?

nuschpl, 2022-10-24:
IMHO The feature of masstorge is important not only from perspective of copying data from/to it. It is also some form of cover channel - one could delay execution of ducky script and pretend it's just a storage, or have an excuse when catched with device connected

PeterHindes, 2023-01-04:
This is a much needed feature for full badusb functionality. The current level of functionality can be achieved with a $5 arduino.

skotopes, 2023-03-13:
Storage subsystem went throw a bunch of optimization. Now we can say that there is nothing else we can do to improve it any more.

We still don't think that current performance is enough for normal work of twin ducky, but community can try to implement it. Could work for some cases.

nuschpl, 2023-04-11:
IMHO we shouldn't directly access Flipper SD card due to Flipperr operator's own safety, but emulate one on some buffer with very limited size suitable for moving config files/powershell scripts back and forth. Then the buffer could be sync with flipper SDcard. Or what's about Flipper internal storage - is 100% occupied by firmware those days ?