flipperdevices/flipperzero-firmware #1043: Flipper File System Showing files beginning with ._

BcsHax, 2022-03-19:
**Describe the bug**
Flipper Built in file system is displaying files with "._" when folders are added in a directory. After opening Sub Ghz files file without "._" the actual capture does not send. 

**To Reproduce**
Steps to reproduce the behavior:
1. Add 1-n folders with multiple files to the Flippers Sub Ghz fold on your SD card
2. From the flipper Desktop press the down arrow to enter the file browser for the
3. Access the folder you added
4. Select a sub ghz file added 

https://user-images.githubusercontent.com/4081137/159099678-53239e62-70bb-4df6-bfb5-09f182649ca6.mov

5. Select "Run the app"
6. Select Send

**Expected behavior**

- Flipper file system does not show hidden files with "_"
- Filler allows user to send Sub Ghz transmission as it would if there were no folders

https://user-images.githubusercontent.com/4081137/159097355-bf229d31-12cc-4531-88c6-1faee01122d4.mov

skotopes, 2022-03-20:
Do I understand your issue correctly?

- you are using mac os 
- you are using card reader to access SD Card
- after copying files to sd card you put it to flipper
- and then in file browser you can see those files (generated by mac os to keep extended attributes in them)

closed 2022-03-23 by skotopes

BcsHax, 2022-03-27:
@skotopes Sorry I wasn't traveling for work. 

Yes, all assumptions are correct. It looks like the new file browser in qFlipper allows you to hide hidden files. Does the flipper now do that too? 

also I'm noticing that I can't play sub ghz files that were not created/recoreded on the flipper if they are in a folder. is that a known issue? if so I can open a new ticket.

skotopes, 2022-03-27:
Flipper at this point is not hiding anything. We are still thinking if it is good idea to add filter or not.

As for subghz yes, please open ticket it's a bug. Just make sure that you using latest firmware.