flipperdevices/flipperzero-firmware #1047: SubGHz Crippled, RX Enabled & TX Disabled

wbraswell, 2022-03-22:
Tag @skotopes & @Skorpionm 

Like many others, I backed the Flipper project with the specific purpose of consolidating multiple remote-control and key-code devices into one, especially the frequencies covered by the sub-gigahertz transceiver.

Imagine my surprise and supreme disappointment when I was finally able to capture the signal of my garage door opener, but then as I tried to transmit the signal I received the error message **"This frequency can only be used for RX in your region"**.

Since my Flipper also seems incapable of reading RFID tags (see bug report), I am basically left with a less-than-convenient infrared remote control for my television.  This is definitely NOT what we all paid money for.  The Flipper Zero page on Kickstarter specifically claims the device will act as a **"Sub-1GHz Transceiver"** that we can **"Use as [a] radio remote"**
[image]

The Kickstarter page goes on to state the following:
**"To communicate with the real world systems, Flipper Zero has a built-in radio module based on TI CC1101 chip. It supports both transmitting and receiving digital signals within the 300-928 MHz frequency range. This is the operating range for a wide class of devices and access control systems such as garage doors remotes, boom barriers, IoT sensors, and remote keyless systems."**
[image]

**"Out of the box, Flipper Zero can emulate remotes for popular garage doors and barriers. You can keep hundreds of remotes in Flipper's memory as well as create a blank remote for the new wireless gate. Just select the right brand of the system in the Flipper menu, register a new key in your garage/barrier receiver, and give it a unique name for easy navigation between your remotes."**
[image]

[https://www.kickstarter.com/projects/flipper-devices/flipper-zero-tamagochi-for-hackers](https://www.kickstarter.com/projects/flipper-devices/flipper-zero-tamagochi-for-hackers)

I understand the Flipper firmware is in a constant state of development - which would explain the current lack of a library of remote control brands & signals - but this seems to be a purposeful crippling of the device which directly conflicts with the publicly-advertised features of Flipper Zero.  Your project raised $4,882,784 from 37,987 backers on Kickstarter, specifically based on this seemingly-false advertising on the Kickstarter page.

According to the US Federal Trade Commission, **"Advertising must tell the truth and not mislead consumers."**
[https://www.ftc.gov/business-guidance/resources/advertising-marketing-internet-rules-road](https://www.ftc.gov/business-guidance/resources/advertising-marketing-internet-rules-road)

Furthermore, **"The FTC periodically joins with other law enforcement agencies to monitor the Internet for potentially false or deceptive online advertising claims.  If your advertisements don't comply with the law, you could face enforcement actions or civil lawsuits."**
[https://www.ftc.gov/business-guidance/resources/advertising-marketing-internet-rules-road#non](https://www.ftc.gov/business-guidance/resources/advertising-marketing-internet-rules-road#non)

Exactly when and how will radio transmission be enabled for the sub-GHz frequencies, as publicly advertised on the Kickstarter page?

skotopes, 2022-03-22:
Wow.

It is actually federal government requirement to comply to FCC regulations and flipper exactly follows them (https://fccid.io/2A2V6-FZ).

Some papers to read: https://www.fcc.gov/wireless/bureau-divisions/technologies-systems-and-innovation-division/rules-regulations-title-47

closed 2022-03-22 by skotopes

wbraswell, 2022-03-22:
Thanks for the links, but that's a whole lot of legalese and technical jargon, which is not exactly easy to parse or understand exactly how it applies to this case.

Is the sub-GHz transmitter disabled for all frequencies, or only some specific frequency ranges?

wbraswell, 2022-03-22:
Where is the list of regions and prohibited frequencies?

How does a Flipper device determine which region it is currently located within?

skotopes, 2022-03-22:
Shipping region is programmed into OTP memory.

I'm not going to advice you to do illegal things, especially I'm not advising to read subghz source code https://github.com/flipperdevices/flipperzero-firmware/blob/dev/firmware/targets/f7/furi_hal/furi_hal_subghz.c#L570 and modify it the way you like it.

Iiridayn, 2022-06-04:
In truth supporting broader than legal TX capabilities in Sub-GHz with the express intention of emulating remote controls you already use and own may be grey area. Certainly the manufacturer cannot and should not ship or support such changes (or they'll be prevented from creating more such devices). However - your hardware and code, your liability. The code is indeed open source and I found it quite pleasant to navigate when I added support for my ceiling fan light.

T0wdup2, 2022-07-07:
Chillout do some more research
 📣ur a potato
 
🎻

silentyak, 2022-07-28:
I barely know how to code and was just browsing and saw this return value https://github.com/flipperdevices/flipperzero-firmware/blob/c7772060657afc8ee5a160ea3ed62246d989c136/firmware/targets/f7/furi_hal/furi_hal_version.c#L252

I also saw this enum value `FuriHalVersionRegionUnknown`

Gambrius, 2022-08-10:
> I barely know how to code and was just browsing and saw this return value
> 
> https://github.com/flipperdevices/flipperzero-firmware/blob/c7772060657afc8ee5a160ea3ed62246d989c136/firmware/targets/f7/furi_hal/furi_hal_version.c#L252
> 
> I also saw this enum value `FuriHalVersionRegionUnknown`

Besides the described regions `R01`, `R02` and `R03` there is a Region `R00` which is named `FuriHalVersionRegionUnknown` within the `OTA -> Board -> Region` Setting.

If anyone knows how to set the value within the OTA memory of the Flipper zero to the mentioned value, it would let you use the whole frequency bandwidth.

Skorpionm, 2022-08-10:
I know. no way!