flipperdevices/flipperzero-firmware #1048: RFID: Unable To Detect Or Read Any Signal

wbraswell, 2022-03-22:
Tag @skotopes & @Skorpionm 

Using a variety of different RFID tags, I have been unsuccessfully attempting to read an RFID signal.

In the **"125 kHz RFID"** menu, I simply select "Read" and I am given the animation of the Flipper dolphin with the caption **"Reading LF RFID"** while the LED rapidly blinks approx 8x red followed by 4x blue, repeating.

No matter where I locate the RFID tag in relation to the Flipper device, I do not receive any change in status on the Flipper LCD screen.  It keeps the same screen and LED blinking indefinitely.  I have tried several different unrelated RFID tags, with the same uneventful outcome.  I especially focused on placing the RFID tags along the **"bottom of Flipper's body"** as specified on the Kickstarter page.
[image]
[https://www.kickstarter.com/projects/flipper-devices/flipper-zero-tamagochi-for-hackers](https://www.kickstarter.com/projects/flipper-devices/flipper-zero-tamagochi-for-hackers)

Was my Flipper device shipped from the factory with the RFID transceiver hardware physically broken, or is this a firmware issue?

(I am running FW 0.50.0.)

skotopes, 2022-03-22:
Do you know exact type of the card?
Are you sure that card you are reading is 125kHz RFID?
Have you tried NFC?

Also please check: https://forum.flipperzero.one/c/125khz-rfid/5

wbraswell, 2022-03-22:
Okay I tried NFC and was able to get a successful read, and I also tried RFID with a different tag and got a successful read, so I will close this issue now, thanks @skotopes !

closed 2022-03-22 by wbraswell