flipperdevices/flipperzero-firmware #1052: Sub-GHz No Longer Sends Via Saved in Dev Branch

MuddledBox, 2022-03-24:
**Describe the bug**
In the latest Dev build, when sending via a saved Sub-GHz signal, it no longer sends. It vibrates, but instantly exits, regardless of length. No green LED blinks as it used to. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open Sub-GHz
2. Click, "Saved"
3. Open any saved Sub-GHz file
4. Click "Send"
5. Notice the signal did not send, and ended abruptly after 100ms (also no green light)

**Expected behavior**
Green light should blink and signal should send. 

**Additional context**
I believe this issue began occurring as a result of the following PR: 
https://github.com/flipperdevices/flipperzero-firmware/pull/1049

@skotopes

skotopes, 2022-03-24:
Thanks for report. Checking.

MuddledBox, 2022-03-24:
Specific Dev Build tested is 3615ffec.

skotopes, 2022-03-24:
> Specific Dev Build tested is [3615ffe](https://github.com/flipperdevices/flipperzero-firmware/commit/3615ffecbe993e20b2346db0192c611f0b0edbb3).

I have a clue on what caused issue. @Skorpionm will check it tomorrow.

skotopes, 2022-03-25:
Fixed in 0.52.1-rc

closed 2022-03-25 by skotopes

MuddledBox, 2022-03-25:
Confirming this is fixed, tested 0.52.2-rc.