flipperdevices/flipperzero-firmware #1059: Cannot exit Mifare Classic screen (0.52.2)

ide, 2022-03-25:
**Describe the bug**
After selecting the option to scan for Mifare Classic ICs, there is no way to exit the UI without a hard reset.

**To Reproduce**
Steps to reproduce the behavior:
1. Choose "NFC" in the main menu
2. Choose "Run special action"
3. Choose "Read Mifare Classic"
4. Cannot exit the next screen

[image]

**Expected behavior**
Pressing the back button should leave the Mifare Classic screen.

**Logs**
I didn't look at the logs. Can provide them if needed.

**Target**
Model FZ.1 running official firmware and bootloader 0.52.2

jman1118, 2022-03-26:
I get a "function requires an SD card with fresh databases" message.

gornekich, 2022-03-28:
@jman1118 you should update firmware with qFlipper

gornekich, 2022-03-28:
@ide thanks for report. Fix just arrived to dev

closed 2022-03-30 by skotopes