flipperdevices/flipperzero-firmware #1067: SubGHz, 300MHz Not Detected By Read & Frequency Analyzer

wbraswell, 2022-03-28:
I am now running the latest firmware v0.52.2.

I am able to use the "Read RAW" functionality in order to record my 300MHz Chamberlain garage door opener, however the "Read" and "Frequency Analyzer" still do not detect any signal.

When I first used my Flipper device out of the box, before upgrading any firmware at all, both the "Read RAW" and "Frequency Analyzer" would successfully detect the 300MHz signal, although the "Read" feature never seemed to detect the signal at all, either then or now.

skotopes, 2022-03-28:
As for Frequency Analyzer it got some limitations: signal must be long enough. If transmitter it not repeating it's payload then we can not detect it.

Read is only for signals that we know how to decode. 

So let's start with FCCID and model.
We will try to find this system and ship it to our RnD office, but we also can start with recorded signal (need signal from couple fobs).

@Skorpionm will help with it

wbraswell, 2022-03-28:
Make: Chamberlain brand
Model: 300MC (garage door remote control)
FCC ID: hbw2281
Frequency: 300MHz

As I said, we at the very least the "Frequency Analyzer" was working well and now it is not.  I can hold down the button on the Chamberlain remote and it will keep it's red light on, which means it is transmitting continuously.  I know this because I was eventually able to get the Frequency Analyzer to properly show 300MHz received after holding down the remote button for several seconds and only when it is positioned perfectly correctly.  The RSSI shows only one bar and barely that.

Out of the box, the Frequency Analyzer was working correctly from any angle or distance or brief period of transmission.  I am using a fresh set of batteries in the remote, just to make sure it's not that.

Did the subGHz sensitivity or squelch algorithm change since v0.43.1?

Skorpionm, 2022-03-29:
many times I asked people who want to add support for new systems to the flipper, throw the necessary data on the forum.  and not one person did it.  I don't have this system yet so I can add it myself.  please do what is written here https://forum.flipperzero.one/t/my-remote-isnt-supported-how-to-add-new-sub-ghz-protocol-in-flipper-zero/2033/2 the record must be made on any sdr receiver with the URH program.  , preliminary correct tuning to the transmitter frequency (there is an analyzer function in the program)

wbraswell, 2022-03-29:
@Skorpionm okay I will follow your directions regarding the capture and submission of 10 button clicks in separate files, as it relates to the "Read" functionality.

The poor performance of the "Frequency Analyzer" functionality seems to be a different issue, and discussed in some depth here:
https://forum.flipperzero.one/t/frequency-analyzer-not-working/1834/22

Skorpionm, 2022-04-14:
https://github.com/flipperdevices/flipperzero-firmware/pull/1108
support for custom frequencies for SubGhz
you can separately set the frequencies for Read, Hopping and frequency analyzer

skotopes, 2022-04-22:
So what do we do? @Skorpionm is protocol supported on our side and issue is fixed?

Skorpionm, 2022-04-22:
I never received a protocol record on SDR, I can’t help but say something about the protocol, now you can add any of the supported frequencies in terms of frequencies

closed 2022-04-25 by skotopes