flipperdevices/flipperzero-firmware #1076: "Burst" wide band frequency keying mode

jinschoi, 2022-03-30:
I am looking at duplicating the protocol for my Honda key fob, for opening doors, unlocking, etc. It appears to be a simple sequence of 25ms bursts every 100ms at either 313.55 or 314.15 MHz (313.85 +-.30 MHz). I would like to be able to send this type of signal, but there is no easy way to do it at this time.

My thought for the easiest way to support this would be to extend the RAW .sub format for OOK to allow for a frequency change immediately after a mark duration:
```
Frequency: 313550000
RAW_Data: 25 -75 25
Frequency: 314150000
RAW_Data: -75 25 -75 25
Frequency: 313550000
...
```

skotopes, 2022-03-30:
Current architecture doesn't allow to do that. Right now timings(CC/AR) are fed to TIM2 directly(DMA) from memory, that makes impossible to change frequency at the same time.

It is however possible: you will need to use lowest level of furi_hal_subghz and directly control gpio_cc1101_g0 from your code. Going to a be little bit challenging though.

skotopes, 2022-03-30:
Can we also have FCCID/mode and/or recorder signal?

skotopes, 2022-04-01:
@Skorpionm what do you think?

Skorpionm, 2022-04-12:
I think that there can’t be a simple sequence of signals, at different frequencies, it’s still some kind of protocol, I still need to hold this key fob in my hands to say more

jinschoi, 2022-04-27:
This is just a regular FSK burst with a rather wide 600 KHz offset. Would it be possible to extend the bandwidth of the raw FM reader to that much?

skotopes, 2022-04-28:
https://fccid.io/KR5V1X/Test-Report/Occupied-Bandwidth-Plot-1913633

it is FSK

Skorpionm, 2022-04-29:
> https://fccid.io/KR5V1X/Test-Report/Occupied-Bandwidth-Plot-1913633
> 
> это ФСК

this shows that Deviation ~ 47.7k, more accurately you need to look at SDR.
Write wav to SDRSharp, there will be a lot less questions

Skorpionm, 2022-04-29:
the transmission frequency +- 30k indirectly confirms that this is FSK. you can also try recording RAW on "FM476" and attach it here, I'll look at it

jinschoi, 2022-04-29:
I attached an IQ recording from an SDR up above in my second comment, looks like this in URH:
[image]

Recording raw at 313.85 with FM476 shows no signal being picked up. Not surprising as 476 I assume means 476KHz bandwidth? It would need 600 KHz.

Skorpionm, 2022-04-29:
look carefully that I asked you to record the frequency range of the Sdrsharp program in order to accurately see the signal parameters.  and not recording at 1 frequency,

jinschoi, 2022-05-22:
Here is a baseband recording using SDRSharp of the above signal. The frequency deviation is 600 KHz (313.55 MHz to 314.15 MHz). Sorry for the long delay.

[SDRSharp_20220522_124911Z_313800000Hz_IQ.wav.zip](https://github.com/flipperdevices/flipperzero-firmware/files/8749438/SDRSharp_20220522_124911Z_313800000Hz_IQ.wav.zip)

Skorpionm, 2022-05-23:
[image]

you have a 2-frequency remote control, with FSK modulation, it looks like the deviation is 27.7 kHz,
the transmission goes alternately on 1 then on 2 frequencies

jinschoi, 2022-05-25:
A detailed spectogram of one of the pulses doesn't show any frequency shifting within the pulse, though.
[image]

Skorpionm, 2022-05-26:
you just have a 2-frequency remote that transmits in turn

closed 2022-09-06 by skotopes