flipperdevices/flipperzero-firmware #1077: Request for 134kHz reading

cshock159, 2022-03-30:
Generic feature request for 134kHz chip reading.

SnowLeopard71, 2022-05-19:
FYI:  http://www.gizmolab.co.za/fdx-b-protocol/
> FDX-B, is an Animal Identification data format covered by ISO 11784/5 and normally operates at 134.2 KHz, which is slightly higher then the 125KHz RFID tags. It uses ASK modulation with BI-Phase encoding.

There is also the specs of the 128-bit stream.

stanjer001, 2022-05-21:
Same here. An ability to help with reading pet chips would be super usable, and saves a lot of time compared to taking them to the vet.

decidedlygray, 2022-05-23:
As a newbie to the project, I was wondering if someone might be able to outline in broad strokes what would be required to implement this?

gorodulin, 2022-07-17:
I have two suspicious looking stray cats in my backyard. They will bite if I try to catch them, but I could scan them to see if they are lost.
Local RFIDs operate on 134.2 KHz

spencerthayer, 2022-07-19:
I've more or less stopped using Flipper, but if and when this gets put into the code repository I'll finally have a use for it.

Catman155, 2022-07-25:
it's being implemented by @DrZlo13 atm

exussum12, 2022-07-27:
Just tried the dev branch for this and it scans well! Great job @DrZlo13 and others involved

hbrylkowski, 2022-07-28:
> Just tried the dev branch for this and it scans well! Great job @DrZlo13 and others involved

can you share which specific branch?

exussum12, 2022-07-28:
https://github.com/mauimauer/flipperzero-firmware/tree/fdxb

databonanza, 2022-08-05:
This would be a great win considering when my cat had a chip injected i was then told that I needed to "register" with some payment service to be able to identify my pet in the future.  I'd like to cut out the middle man, my cat has a chip in it.... I should be able to read it just like anyone else should be able to read it.

exussum12, 2022-08-05:
The other service is still needed. The chip gives a number out. The number to pet info lookup is provided by a service similar to the one you are talking about.

databonanza, 2022-08-05:
@exussum12 - wouldn't we just make a public registry?  I don't see why you'd have to pay a service to do this.  Enable the public to scan their own pets, provide a way for them to publicly register the numbers/contact info.  Seems like the right thing to do vs. catering to a private "for pay" registry.

decidedlygray, 2022-08-05:
Couple issues with that I think:

- there isn't just 1 microchip provider, so there are actually multiple registries
- even if it's a public registry, someone has to pay to create and subsequently maintain the web app and API

More info: https://www.aaha.org/your-pet/pet-microchip-lookup/microchip-search/

databonanza, 2022-08-05:
I'm not sure I follow.  Github seems pretty free to me.  Why couldn't a repo here do the job?  I'm not attempting to be sly here, I'm just wondering why a paid service would be needed at all.  Yes "someone" pays for github.com to stay online, but it's not you or me.

phptanker, 2022-08-05:
I don't really care about being able to read from any pet database. I just want to scan my cat's RFID chips so I can make cool automations, for example count how many times they visit the bathroom box every day.

exussum12, 2022-08-05:
@databonanza time is not free either. Hosting on GitHub would mean managing pull requests.

exussum12, 2022-08-05:
@phptanker this current version wouldn't work well. While I can scan my dog it's not trivial to find. I think you would have better luck with a dedicated scanner

SnowLeopard71, 2022-08-05:
>  I needed to "register" with some payment service to be able to identify my pet in the future.

@databonanza  Weren't you referred to a specific service to register with?    Normally, the registration fee is included in the cost of the chip and you don't have to pay extra.

exussum12, 2022-08-05:
I know I did not pay for it originally, but to change my info (when moving house) I had to pay a registration fee of £15 from memory.

databonanza, 2022-08-05:
> @databonanza time is not free either. Hosting on GitHub would mean managing pull requests.

Preaching to the choir.  My time is VERY valuable..... according to me and my employer.  I'm trying to make it so valuable that I don't have to use it making money anymore.  That is top value.

> > I needed to "register" with some payment service to be able to identify my pet in the future.
> 
> @databonanza Weren't you referred to a specific service to register with? Normally, the registration fee is included in the cost of the chip and you don't have to pay extra.

Yes, they wanted some yearly fee, it was like a subscription service.  I think a "free service" could easily be made w/ private donations.  If I were a multimillionaire I would gladly fund something like this out of pocket for the betterment of man.

klimeryk, 2022-08-21:
> it's being implemented by @DrZlo13 atm

For those that want to follow the PR: https://github.com/flipperdevices/flipperzero-firmware/pull/1601

ender003, 2022-08-23:
To add to the request - please also include the ability to read/emulate ISO17785 HDX half-duplex.

exussum12, 2022-08-27:
This can be closed now, it was added in the latest stable firmware

Taby1993, 2022-08-27:
Can flipper change number microchip?

skotopes, 2022-09-06:
Check latest release and if you found unsupported tag then please report exact vendor/model

closed 2022-09-06 by skotopes

Taby1993, 2022-09-06:
Just this question, Can he change the number on the chip? Please answer to me.

Closed #1077<https://github.com/flipperdevices/flipperzero-firmware/issues/1077> as completed.

skotopes, 2022-09-06:
Depends on what chip is used: if it is t5577 then yes.

Taby1993, 2022-09-06:
Fdx B iso 11784/5 134.2 Khz?

skotopes, 2022-09-06:
@DrZlo13

DrZlo13, 2022-09-07:
Our RFID scheme is not designed to receive HDX protocol. The rest protocols (FDX-A, FDX-B) mentioned in this issue is done.

Taby1993, 2022-09-07:
Ok thanks