flipperdevices/flipperzero-firmware #1081: [NFC] Detecting Ultralight will run indefinitely when scanning

jman1118, 2022-04-01:
Bug - Ultralight script will run in infinite loop when scanning for tag

Steps to reproduce the behavior:
1. Switch on
2. Select NFC
3. Read Card
4. More
5. Run compatible app

Expected Behavior:
Flipper should time out after X seconds if no card detected or return a result

skotopes, 2022-04-04:
Looks abnormal. Can you also switch "Settings->System->Log Level" to debug and  provide console log?

gornekich, 2022-04-20:
Ultralight cards can protect memory area with password. In this case you have to know password to authenticate to protected memory. This functionality is not implemented in flipper so far.
However, I can say that this is the reason why you can't read card if you provide console logs

jlaughter, 2022-05-13:
Here's the console log for this - repeats indefinitely until cancelled:

10970315 [D][NfcWorker]: Found Mifare Ultralight tag. Start reading
10970316 [D][MfUltralight]: Reading version
10970321 [D][MfUltralight]: Reading signature
10970373 [D][FuriHalNfc]: Timeout during data exchange
10970375 [D][MfUltralight]: Failed redaing signature
10970377 [D][MfUltralight]: Reading counters
10970430 [D][FuriHalNfc]: Timeout during data exchange
10970431 [D][MfUltralight]: Failed to read 0 counter
10970433 [D][MfUltralight]: Reading tearing flags
10970486 [D][FuriHalNfc]: Timeout during data exchange
10970490 [D][MfUltralight]: Reading pages 0 - 3
10970542 [D][FuriHalNfc]: Timeout during data exchange
10970544 [D][MfUltralight]: Failed to read pages 0 - 3
10970546 [D][NfcWorker]: Failed reading Mifare Ultralight

GMMan, 2022-06-15:
Suggest trying to read this with NXP NFC TagInfo to see if there are any specific configurations on the tag. Most notably, whether the entire tag can be read without authenticating.

skotopes, 2022-09-06:
Please check latest Release and reopen if issue persist

closed 2022-09-06 by skotopes