flipperdevices/flipperzero-firmware #1084: Feature request: Sniff MF Classic keys from readers.

djsime1, 2022-04-04:
**Is your feature request related to a problem? Please describe.**
While the Mifare Classic key database covers 95% of scenario when reading tags, there are some cases where it doesn't.

**Describe the solution you'd like**
I would like to have the ability to sniff sector keys as sent by a tag reader. In concept, Flipper would emulate a tag and capture access keys when they are sent. After the keys are intercepted, the user would be prompted to save them to a secondary database file that is used to read future tags in conjunction with the existing key database.

**Describe alternatives you've considered**
While not exactly an alternative to the point of this request, it would be nice to see if users could input custom access keys on device (as part of the aforementioned secondary database).

**Additional context**
Understandably, this isn't possible until Mifare Classic emulation is completed. Even then, I don't know how low level Flipper interacts with raw NFC transmission or if it could sniff keys in the first place. In any case, this request is far from a priority.

skotopes, 2022-04-07:
We are working on MiFare classic, we'll publish update after finish research

micsen, 2022-04-22:
This could be relatively simple to implement but you will most likely need to take the nonce's from the reader to another device to calculate the key as that is quite resource intensive.

skotopes, 2022-05-30:
In theory yes

djsime1, 2022-06-10:
After refamiliarizing myself with the Mifare Classic spec document, it seems like only one key is almost guaranteed to be transmitted by the reader. From there it depends how smart the reader is to send the rest. Since the MFC subsystem already uses transparent mode for operation, I don't see anything stopping the Flipper from attempting to make the reader authenticate at least once (besides the timing issue). I also remember reading somewhere that older MFC cards don't enforce proximity/timing checks, meaning a middleman attack could be possible too (but even I think this is too far out of scope, and might be pushing into the hacking territory). So I guess that leaves this request in somewhat of a gray zone.

simkard69, 2022-07-28:
> timing issue

As I'm new to Flipper Zero (just received it yesterday), I was wondering about this "timing issue".
It's not the first time I'm hearing about it and I'm worried that it is a H/W related problem rather than a S/W one.
Am I correct ?

djsime1, 2022-07-28:
Quote from [this FAQ](https://github.com/djsime1/awesome-flipperzero/blob/main/FAQ.md#why-isnt-mifare-classic-emulation-working):
> Flipper's NFC chip doesn't have hardware support for Mifare Classic, so it's been offloaded to the CPU. However, the CPU's clock cycle can't conform to the exact (and strict) timings that Mifare Classics communicate with. This means that some readers will respond to emulation, while others won't. This can not be fixed with firmware.

You are correct; but I've experienced that most dedicated reader hardware works, while smartphones usually are the ones that cause issues.

Linzdigr, 2022-07-31:
@djsime1 VIGIK readers that are quite widely used here in France denies the current MFC 1k emulation (dev https://github.com/flipperdevices/flipperzero-firmware/commit/c40e8811d68e9f4b8f603ae5d5826b814521014d), unfortunately.

skotopes, 2022-09-06:
Please check latest Release.

closed 2022-09-06 by skotopes