flipperdevices/flipperzero-firmware #1085: Sub-GHz: hopping v2

otanim, 2022-04-04:
**Is your feature request related to a problem? Please describe.**
no

**Describe the solution you'd like**
For Sub-GHz: Hopping: allow to be able to select desired frequencies/modulations.

**Describe alternatives you've considered**
Well, it can be hard-coded inside the source code directly but would be great to be able to do this directly from the firmware's UI.

**Additional context**
"This will help to target and read required data much quicker".

skotopes, 2022-04-07:
Yep, we working on adding custom frequencies.

otanim, 2022-04-10:
@skotopes :pray: danke

Skorpionm, 2022-04-14:
https://github.com/flipperdevices/flipperzero-firmware/pull/1108
support for custom frequencies for SubGhz

otanim, 2022-04-16:
@skotopes / @Skorpionm, a suggestion from my side for https://github.com/flipperdevices/flipperzero-firmware/pull/1108 PR:

**PR's change:**
- Can this be integrated inside the UI. Otherwise, be documented with example files?

**In general:**
- If any new feature has been added to the firmware, it either has to be UX friend in such a way that it would be operable using internal UI, elsewise, if the functionality is too complex to integrate inside the UI, then do proper documentation, otherwise, the new feature will be in production's release but won't be visible unless someone digs into PR's descriptions.

Skorpionm, 2022-04-17:
yes, it is logical, it will be described in the documentation, it is currently at the testing stage, but everything is simple there, this is a list of frequencies that you need, there can be as many (within reasonable limits) as many of them, sorting is not important, and only frequencies available cc1101 will work,  others he will discard

skotopes, 2022-04-22:
included in 0.55.0

closed 2022-04-22 by skotopes