flipperdevices/flipperzero-firmware #1092: Saved Mifare Classic Access Difficulty

OMGNapples, 2022-04-05:
**Describe the bug**
After saving a Mifare Classic card, if I attempt to emulate it from NFC's "saved cards", there is no option.
Temporary workaround: accessing the saved NFC cards from the favorites/saved menu will allow you to Run in app

**To Reproduce**
Steps to reproduce the behavior:
1. Save a Mifare Classic card
2. Return to NFC app screen
3. Select "Saved cards"
4. Select your Mifare Classic
5. A blank screen appears temporarily
6. "Edit UID and name", "Delete", and "Info" options exist
7. Return to home menu
8. Hit down arrow
9. Navigate to NFC saved Mifare Classic and select
10. Select Run in app

**Expected behavior**
I expected to have a function to emulate the card from the NFC app, not just the quick access

**Additional context**
Not sure if this is relevant, but I was unable to fully capture and emulate the card. I am missing a few keys each time I read it. I have not been able to successfully complete the Mifare Classic workflow yet, so I don't know if this would happen on a working capture. 

I'm using this dictionary: 
https://raw.githubusercontent.com/flipperdevices/flipperzero-firmware/dev/assets/resources/nfc/assets/mf_classic_dict.nfc

gornekich, 2022-04-20:
We hide emulate option intentionally since we have not implemented Mifare Classic emulation so far. You can only emulate UID from archive. Mifare Classic emulation is under development

skotopes, 2022-09-06:
Check latest Release and reopen if issue persist.

closed 2022-09-06 by skotopes