flipperdevices/flipperzero-firmware #1100: Bluetooth Remote corrupts radio firmware

ase1590, 2022-04-07:
When entering the bluetooth remote app on Flipper, waiting for it to pair, then using it, it may cause QFlipper to detect the radio firmware as corrupt, as well as in some cases locking up the Flipper entirely. 

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on...
2. enter Plugins -> Bluetooth Remote
3. wait for pairing to complete
4. press a few buttons on the flipper
5. exit the app on the flipper
6. re-enter the app on the flipper
7. wait for pairing again
8. press a few buttons
9. exit the app
10. plug in the Flipper, and launch QFlipper
11. the Radio firmware will now be detected as corrupt.  

**Additional context**
behavior occurs both on firmware 0.53.0 as well as the latest compiled dev brasnch firmware.

skotopes, 2022-04-07:
Hi. Thanks for report.

skotopes, 2022-07-10:
Fixed in 0.5x

closed 2022-07-10 by skotopes