flipperdevices/flipperzero-firmware #1110: ONN remote code for database

fennectech, 2022-04-13:
https://flpr.app/s/#path=ext/infrared/Onn_tv.ir&Filetype=IR+signals+file&Version=1&name=Pwr&type=raw&frequency=38000&duty_cycle=0.330000&data=4035+3994+517+1982+517+1982+517+1981+518+1981+518+1006+497+1001+492+1982+517+1007+496+1978+521+1004+489+1983+516+1009+494+1003+490+1009+494+1003+490+1008+495+1978+521+1977+522+1003+490+1982+517+1007+496+1977+522+976+517+1983+516&name=Mute&type=raw&frequency=38000&duty_cycle=0.330000&data=4042+3988+522+1977+521+1978+520+1979+519+1980+518+1006+487+985+518+2008+490+1982+516+1985+513+1986+512+1986+512+1987+522+1002+491+1008+495+1003+490+1009+494+1978+520+1979+519+978+515+1010+493+979+514+1011+492+1006+487+1012+491&name=V_up&type=raw&frequency=38000&duty_cycle=0.330000&data=4033+3995+515+1984+514+1985+513+1985+513+1986+512+1011+492+980+523+1976+522+976+516+1984+514+1984+514+1985+513+1986+512+985+518+981+522+976+517+1008+495+2004+494+1978+520+978+514+1984+514+1010+493+979+514+984+519+981+522&name=V_dn&type=raw&frequency=38000&duty_cycle=0.330000&data=4041+3988+521+1979+519+1980+518+1981+517+1982+516+982+521+1004+488+1984+514+1011+492+1982+516+1984+514+1985+513+984+519+979+514+985+518+981+522+1003+490+1982+516+2010+488+983+520+1980+518+980+523+976+516+982+521+1978+520&name=Ch_up&type=raw&frequency=38000&duty_cycle=0.330000&data=4041+4014+495+2004+494+2004+494+1979+519+1979+519+979+513+985+518+1981+517+981+522+1978+520+1979+519+979+513+1986+522+1002+490+981+522+977+515+1009+494+1978+520+1980+518+1006+497+1976+522+1002+490+1008+495+1978+520+979+513&name=Ch_dn&type=raw&frequency=38000&duty_cycle=0.330000&data=4042+4013+496+2003+495+1978+520+1979+519+1980+518+980+523+975+517+1982+516+1008+495+2006+492+1981+517+981+522+1003+489+1009+494+1004+488+984+519+979+513+1986+522+1977+521+1003+489+1984+514+1010+493+1005+487+1986+522+1977+521

Here is a remote code for an ONN tv  you can add it to the database.   I minimized the packet lengths as much i could. Seems 51 bytes is the minimum for it to work

fennectech, 2022-04-13:
https://flpr.app/s#path=ext/infrared/remote.ir&Filetype=IR+signals+file&Version=1&name=V_up&type=raw&frequency=38000&duty_cycle=0.330000&data=193+899+220+1734+225+726+227+2825+223+1280+223+1280+223+1149+219+2701+222+12626+230+910+199+1463+227+729+225+781+204+727+226+2144+220+763+201+751+254&name=V_dn&type=raw&frequency=38000&duty_cycle=0.330000&data=198+921+198+1730+229+722+221+2831+227+1275+228+1275+228+1144+224+2697+226+12622+224+916+203+1326+198+733+221+758+226+731+222+2305+225+732+221+758+247&name=Ch_up&type=raw&frequency=38000&duty_cycle=0.330000&data=202+890+229+1725+224+753+200+2826+222+1281+222+1281+222+1150+218+2703+220+12628+228+912+197+1071+204+726+227+778+196+735+229+2587+222+760+194+759+257&name=Ch_dn&type=raw&frequency=38000&duty_cycle=0.330000&data=198+920+199+1729+220+732+221+2831+227+1275+228+1275+228+1144+224+2697+226+12623+222+918+201+908+222+735+229+750+224+733+220+2728+226+730+223+755+250&name=Mute&type=raw&frequency=38000&duty_cycle=0.330000&data=197+921+198+1730+229+722+221+2831+227+1277+226+1277+226+1172+196+2699+224+12625+229+884+225+1174+225+732+221+757+227+729+224+2435+229+728+225+753+252

Here are some dudes for my cable box.

fennectech, 2022-04-15:
https://flpr.app/s#path=ext/infrared/remote.ir&Filetype=IR+signals+file&Version=1&name=V_up&type=parsed&protocol=Samsung32&address=07+00+00+00&command=07+00+00+00&name=V_dn&type=parsed&protocol=Samsung32&address=07+00+00+00&command=0B+00+00+00   Here are some Volume buttons for the vast majority of Samsung TVs in the last 15 years

skotopes, 2022-12-01:
@fennectech would you like it to be a part of universal remote?

fennectech, 2022-12-01:
yes.

closed 2023-03-13 by skotopes

fennectech, 2023-07-14:
```Filetype: IR signals file
Version: 1
# 
name: Vol -
type: parsed
protocol: NECext
address: EA C7 00 00
command: 10 EF 00 00
# 
name: Vol +
type: parsed
protocol: NECext
address: EA C7 00 00
command: 0F F0 00 00
# 
name: Mute
type: parsed
protocol: NECext
address: EA C7 00 00
command: 20 DF 00 00
# 
name: Power
type: parsed
protocol: NECext
address: EA C7 00 00
command: 17 E8 00 00
```

Here are some buttons for ONN brand roku TVs

fennectech, 2023-07-14:
@skotopes   Here are some remote codes that arent in the universal remote DB