flipperdevices/flipperzero-firmware #1111: Unable to add files to qflipper ir or subghz

tenyujump, 2022-04-14:
When trying to add any file to the ir or subghz through qflipper I get an operation error. 

[RPC] (129) Storage List @/ext START
[RPC] (129) Storage List @/ext SUCCESS
[RPC] (130) Storage List @/ext/infrared START
[RPC] (130) Storage List @/ext/infrared SUCCESS
[RPC] (131) Storage Write @/ext/infrared/ START
[RPC] (131) Storage Write @/ext/infrared/ ERROR: Failed to open file for reading: No file name specified

skotopes, 2022-04-14:
@tenyujump can you also provide versions of qFlipper and firmware that you were using? Also can you provide full qflipper log?

skotopes, 2022-04-22:
Please reopen issue if it persist with information on FW/qFlipper version.

closed 2022-04-22 by skotopes