flipperdevices/flipperzero-firmware #1120: Sub-GHz doesn't send via folders

jiserra, 2022-04-14:
**Describe the bug**
If a .sub file is located inside a folder and not in the root `subghz` (e.g. `SD Card/subghz/coolsubs/test.sub`) then it doesn't send the signal. It vibrates, but instantly exits, regardless of length. No green LED blinks.

**To Reproduce**
1. Press down arrow button on Home
2. Navigate to the Sub-GHz section
3. Select a folder and navigate inside the folder
4. Select a .sub file
5. Click the "Run in app" menu
6. Click "Send"

**Expected behavior**
Green light should blink and signal should send.

**Actual behavior**
It vibrates for a millisecond, the wave is shown for a millisecond and no signal is sent.

**Additional context**
To my knowledge, this only happens with the .sub files. All the other sections (infrared, NFC, etc) work as intended.

skotopes, 2022-04-14:
@Skorpionm take a look please

Skorpionm, 2022-04-16:
https://github.com/flipperdevices/flipperzero-firmware/pull/1125
fix

closed 2022-04-22 by skotopes