flipperdevices/flipperzero-firmware #1141: Partial transmission of RAW Subghz sub file

samuel, 2022-04-21:
**Describe the bug**
A RAW file I created seems to be only partially transmitted based on the signal I capture using an RTL-SDR. According to the log the entire file was read (based on the on/off total durations in the log):
```
>>> x = [int(k) for k in "620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -1920 1860 -620 1860 -620 620 -1860 620 -1860 620 -1860 620 -1860 1860 -620 1860 -620 1860 -620 620 -1860 620 -1860 1860 -620 1860 -620 1860 -620 1860 -620 620 -9920 1860 -620 1860 -620 620 -1860 620 -1860 620 -1860 620 -1860 1860 -620 1860 -620 1860 -620 620 -1860 620 -1860 1860 -620 1860 -620 1860 -620 1860 -620 620 -9920".split(" ")]
>>> sum(k for k in x if k > 0)
57040
>>> sum(-k for k in x if k < 0)
69500
```

Compared to `26207457 [D][FuriHalSubGhz]: Async TX Radio stats: on 57040us, off 69500us, DutyCycle: 45%`

test.sub:
```
Filetype: Flipper SubGhz RAW File
Version: 1
Frequency: 433800000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: RAW
RAW_Data: 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -620 620 -1920 1860 -620 1860 -620 620 -1860 620 -1860 620 -1860 620 -1860 1860 -620 1860 -620 1860 -620 620 -1860 620 -1860 1860 -620 1860 -620 1860 -620 1860 -620 620 -9920 1860 -620 1860 -620 620 -1860 620 -1860 620 -1860 620 -1860 1860 -620 1860 -620 1860 -620 620 -1860 620 -1860 1860 -620 1860 -620 1860 -620 1860 -620 620 -9920
```

Received signal:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Copy the test.sub file to the FlipperZero
2. Transmit the raw file
3. Record the transmitted signal using an SDR device
4. Compare against the expected on/off durations

**Expected behavior**
The full signal to have been sent rather than partial.

**Logs**
```
26207310 [I][SubGhzFileEncoderWorker]: Worker start
26207337 [I][SubGhzFileEncoderWorker]: Start transmission
26207354 [I][SubGhzFileEncoderWorker]: End read file
26207414 [I][SubGhzFileEncoderWorker]: Stop transmission
26207417 [D][DolphinState]: icounter 60, butthurt 0
26207457 [D][FuriHalSubGhz]: Async TX Radio stats: on 57040us, off 69500us, DutyCycle: 45%
26207507 [I][SubGhzFileEncoderWorker]: Worker stop
```

**Target**
Specify the target

**Additional context**
Using the latest firmware (0.55.1). Capture signal using an RTL-SDR and a HackRF to verify that it's not the receiving side showing the issue.

Skorpionm, 2022-04-21:
I will check tomorrow

Skorpionm, 2022-04-22:
Fix https://github.com/flipperdevices/flipperzero-firmware/pull/1146
[image]

if the last duration is very important, then either end with a high level or make an additional period, or simply repeat the send as a whole. anyway, 1 time is not enough for the receiver

[image]

skotopes, 2022-04-22:
Fixed in dev, will be included in 0.56.1

closed 2022-04-22 by skotopes