flipperdevices/flipperzero-firmware #1150: Cancelling a raw subghz transmission freezes the Flipper.

samuel, 2022-04-22:
**Describe the bug**
Cancelling a raw subghz transmission (hitting back before send completes) freezes the Flipper. My guess this is likely related to the fix in #1146 (thanks for that fix btw, it works great now to transmit the full RAW file).

**To Reproduce**
Steps to reproduce the behavior:
1. Load a somewhat large sub file so that there's time to exit before completion.
2. Press the back/exit button.
3. Interface is now frozen.

**Expected behavior**
Hitting back should stop the transmit.

**Logs**

```
>: log
Press CTRL+C to stop...
174335 [I][SubGhzFileEncoderWorker]: Worker start
174358 [I][SubGhzFileEncoderWorker]: Start transmission
174438 [D][DolphinState]: icounter 60, butthurt 0
174832 [I][SubGhzFileEncoderWorker]: End read file
175017 [D][FuriHalSubGhz]: Async TX Radio stats: on 353345us, off 331823us, DutyCycle: 52%
```
and then nothing.

Doesn't show the `End transmission` line as normally would if allowed to complete the transfer.

**Additional context**
Using a build of the latest `dev` branch at commit 7ed5b9053a526119fe02ba39056a4c1bbdd0c1b7

skotopes, 2022-04-22:
Thanks, working on fix.

Skorpionm, 2022-04-22:
Yes, maybe I'm now waiting for the end of the transfer, and somehow I didn’t think about how the forced cancellation will work now.  fix it!

Skorpionm, 2022-04-22:
https://github.com/flipperdevices/flipperzero-firmware/pull/1152 Fix

closed 2022-04-25 by skotopes