flipperdevices/flipperzero-firmware #1154: Folders in saved signals/cards

Moonman0922, 2022-04-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I tried to add a subfolder of door codes within my saved sub-GHz signals. The Flipper doesn't recognize the folder.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I would like to see the ability to place signals and cards in folders to sort them.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
A tagging system might also work.
**Additional context**
Add any other context or screenshots about the feature request here.

ZeroXR-13mg, 2022-04-23:
Seconding this. I would love the ability to have folders, because I would love to be able to sort my Amiibo scans by folders.

It's definitely a bit messy with just having everything dumped into their respective directories (sub-GHz, NFC, IR, etc)

Skorpionm, 2022-04-23:
already now you can arrange signals for SubGhz in any folders and gifts, but for now it is possible to download them only through the archive (down button on the main screen)

Moonman0922, 2022-04-23:
> already now you can lay out signals for SubGhz in any folders and subfolders, but for now it is possible to load them only through the archive (down button on the main screen)

I did find this after I made the feature request. It would be really nice to be able to see them in the menu too. Also. QFlipper doesn't allow moving existing files into folders. Just ones uploaded from a computer. Allowing drag and drop would be very cool.

Skorpionm, 2022-04-23:
You just thought, and we have already done.  working ahead of the curve

Moonman0922, 2022-04-23:
You all are great! This is some real 5 star service!!

closed 2022-04-25 by skotopes