flipperdevices/flipperzero-firmware #1157: subhz NICE Smilo TX leads to sys halt

J-Run, 2022-04-23:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Put one of [attached](https://github.com/flipperdevices/flipperzero-firmware/files/8548588/subghz.zip) `.sub` files to `/ext/subhz/`
2. go to `Sub-GHz` -> `Saved` then select test case entity
3. make sure that mfcode is loaded and packet deconstructed correctly (all the fields are in place and seems ok)
4. press `Send`
5. Flipper is stuck to the dead until hard reset (left+back)

**Expected behavior**
Current state packed to raw bytes, sent to the air and then new values  calculated depending on counter (like it works with other MF's)

**Logs**
Nothing interesting at CLI with debug trace enabled

**Target**
f7 dev 9351076c

Skorpionm, 2022-04-24:
nice smilo is Keelog encryption, we only support DoorHan transmission.  all other systems on Keelog are not supported for transfer, and you do not need to change files manually :)

J-Run, 2022-04-24:
Thx! got it! 
"it’s not a bug, it’s a feature"

closed 2022-04-24 by J-Run