flipperdevices/flipperzero-firmware #1158: [Power] Option to reduce charging limit for increased battery longevity

digitalcircuit, 2022-04-24:
**Is your feature request related to a problem? Please describe.**
From what I understand, lithium batteries wear out over time, and [using and storing lithium batteries at full charge (e.g. 4.2v) can result in the battery wearing out faster]( https://batteryuniversity.com/article/bu-808-how-to-prolong-lithium-based-batteries ).

Since the Flipper Zero's battery is not easily replaceable (requires disassembly), I'd like to take steps to ensure it degrades as little as possible when I don't need the full battery capacity.

**Describe the solution you'd like**
Provide an option in `Settings` → `Power` to reduce the maximum allowed battery charge for sake of battery longevity.

For example, `Battery charge limit: 100%`, adjustable down to `60%` or so in 10% decrements.

Alternatively, just provide a few charging profiles (`Full Charge`, `Balanced`, `Max Lifespan`) corresponding to 100%, 80%, and 60% or so (see System76's UI in additional context).

*Implementation details:*
Flipper's firmware currently can [pause charging with `furi_hal_power_suppress_charge_enter()` and `furi_hal_power_suppress_charge_exit()`](https://github.com/flipperdevices/flipperzero-firmware/blob/9351076c89cad62e8a33fd2e809e7e81eaa37fca/firmware/targets/f7/furi_hal/furi_hal_power.c#L409-L433 ).

Alternatively, [it appears there may be a way to adjust the bq25896 regulator's voltage limit via `VREG` in `REG06`](https://github.com/flipperdevices/flipperzero-firmware/blob/2f3ea9494e577dd4c0f0dd2bfa18871476f96c38/lib/drivers/bq25896_reg.h#L101 ).  If feasible, lowering that may avoid requiring having the Flipper firmware actively toggle suppressing the charging circuitry.

Either way, the Desktop UI's battery indicator should probably show that the battery charge is limited to avoid potentially confusing folks.

**Describe alternatives you've considered**
* Simply unplugging the Flipper Zero before it fully charges
  * Pros: free, no development effort required
  * Cons: easy to forget, doesn't work when plugged into a computer for USB features
* Hardware charge limiting device
  * Pros: can be automated, no changes to Flipper requires
  * Cons: requires power measuring or coordination with Flipper firmware, needs an extra dongle, may interfere with USB usage
* Letting Flipper charge to 100% / not caring
  * Pros: easiest, no changes or effort required, battery replacement is at least possible (screws etc, no glue in assembly)
  * Cons: may wear out the battery faster through higher charge cycles

**Additional context**
There's ongoing research around battery charge thresholds and various devices and software exists to control this on other platforms.  It'd be nice if Flipper could integrate this without needing a third-party workaround.

* [Battery University article on prolonging lithium battery lifespan](https://batteryuniversity.com/article/bu-808-how-to-prolong-lithium-based-batteries )
* [ACC (Advanced Charge Controller) for rooted Android devices](https://github.com/VR-25/acc/#readme )
* [Chargie hardware charge limiting device](https://chargie.org/ ) for phones and USB devices
* System76's [battery charging threshold documentation](https://support.system76.com/articles/laptop-battery-thresholds/ ) and [UI implementation in GNOME Control Center](https://github.com/pop-os/gnome-control-center/pull/140#issuecomment-763757743 )
* Dell provides a charge threshold setting in their laptop UEFI firmware
* Electric vehicles (Tesla, etc) may have settings for "daily trip" versus "long drive"

skotopes, 2022-04-25:
Hi.

It is possible to limit target battery voltage(with charger settings) and make another set of params for gauge(or leave gauge with current params).
But in my opinion it's not worth it, but you are welcome to implement it. Shouldn't be very difficult.

digitalcircuit, 2022-04-25:
Thank you for your feedback!

I understand it not being a priority.  Knowing you're not opposed to accepting a pull request of sufficient quality, this might be a good opportunity for me to try to learn a manageable subset of every layer in the Flipper firmware (UI, abstractions, hardware control, etc).

Meanwhile, if anyone else sees this open issue, feel free to experiment - it likely will be a long while before I have the time to dig into this.

digitalcircuit, 2022-04-30:
### Minor design update
I've been pondering how to implement this in a way that reduces support requests from folks turning on the feature and forgetting about it.

So far, I've drafted the following change to the battery icon design when charge limiting is enabled:
* The "limited" portion of charge will be grayed out in a checkerboard pattern

[image]

In the above example, the charge is limited to 80% (*i.e. 20% of the battery is "grayed out"*).  This would be drawn inside [`power_draw_battery_callback()` alongside the call to `canvas_draw_box()`](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/applications/power/power_service/power.c#L11-L29) whenever charge limiting is turned on.

I'll dig into the code side when I have time to set up the build environment and tinker - I'm expecting to change the charging profile (*translating the percentage limit into a reduced battery voltage limit*), but not change the charge gauge.

digitalcircuit, 2022-07-06:
# Feedback request - preferred charge limit method?

There's two reasonable methods I've managed to come up with to implement battery charge limiting:

1.  Set charge voltage for [`bq25896`](https://www.ti.com/lit/ds/symlink/bq25896.pdf) and adjust charge voltage in [`bq27220`](https://www.ti.com/lit/ds/symlink/bq27220.pdf) to know when charging is done
    * ℹ Works well with setting charge limit in terms of battery charging voltage (requires Li-Ion knowledge)
    * ➕ After configuration, doesn't need the Flipper Zero firmware to intervene
    * ➕ Unlikely to result in desync issues (where Flipper firmware thinks charging is paused when it isn't)
    * ➖ Specific to the current ICs used (more work if future production batches need to change parts)
    * ➖ More complicated implementation, especially with changing charge limit at runtime
    * ➖ Possible risk of overcharging if a wrong/corrupted voltage charge limit is sent

2.  Pause charging in Flipper firmware upon reaching limit, adjust code to detect when charge is complete
	* ℹ Works well with setting charge limit in terms of percentage
    * ➕ Simpler logic with precise user-facing charge percentage limit (set to 60% = 60% is shown in UI)
    * ➕ Doesn't involve mapping voltages to percentages to determine what percent charge equals the charge limit
    * ➕ Fairly generic across battery charge and gauge hardware
    * ➖ More background work for Flipper Zero firmware (un/pausing charging), potentially more code
    * ➖ Will need to be careful to ensure that the actual charge state remains in sync with Flipper firmware

**Flipper developer team - what would you prefer?**

I'm leaning towards option #**2** given the complexity of #**1** involving coordinating the charge limit via both the charging chip and the battery gauge chip, but I'm happy to try that option if preferred.  I'm open to other methods, too.

It's quite possible I overlooked pros and cons in either solution!

## Additional context - Linux battery charge limit
The Linux kernel [battery charge limit interface operates on start/stop percentages via `charge_control_start_threshold` and `charge_control_end_threshold`](https://www.kernel.org/doc/Documentation/ABI/testing/sysfs-class-power ) (the exact hardware implementation isn't specified).

System76 have implemented battery charge limiting in their open embedded controller firmware for their Linux laptops [using battery charge percentage rather than directly setting the battery charge voltage](https://github.com/system76/ec/blob/a70ea4558c84ec8b76faaf4aa7c11165c2b1bf38/src/board/system76/common/battery.c#L51-L77 ).

This most closely matches option #**2** - software/firmware charge limiting instead of hardware (IC) limiting.

# Details for battery charge/gauge chips
**NOTE:** This does not apply for method #**2**, where the Flipper Zero firmware intervenes to limit charging below 100%.

<details><summary><strong>ℹ Click/tap to show all the details about the bq25896 and bq27220</strong></summary>

I've finally gotten a local firmware build set up using Podman instead of Docker (no daemon running as root), so I've looked into this more…

## `bq25896` - battery charging control

This chip controls when and how much to charge the Flipper Zero's internal battery, and provides the power-off shipping mode, short-circuit protection, and other functionality.

Flipper currently interacts with this chip to pause charging when using the SubGhz app via `furi_hal_power_suppress_charge_enter()` and `furi_hal_power_suppress_charge_exit()`.

Digging into [the datasheet for Texas Instrument's `bq25896` ("I²C Controlled Single Cell 3-A Fast Charger with MaxCharge Technology for High Input Voltage and Adjustable Voltage USB On-the-Go Boost Mode")](https://www.ti.com/lit/ds/symlink/bq25896.pdf), section `9.4.7 REG06` on page 38 describes the possible Charge Voltage Limits:

### Datasheet on Charge Voltage Limit

#### Table 12. REG06 (partial)
Bit | Field | Type | Reset | Description
----|--------|--------|----------|----------------
7  | `VREG[5]` | R/W | by `REG_RST` & Watchdog | 512mV
6  | `VREG[4]` | R/W | by `REG_RST` & Watchdog | 256mV
5  | `VREG[3]` | R/W | by `REG_RST` & Watchdog | 128mV
4  | `VREG[2]` | R/W | by `REG_RST` & Watchdog | 64mV
3  | `VREG[1]` | R/W | by `REG_RST` & Watchdog | 32mV
2  | `VREG[0]` | R/W | by `REG_RST` & Watchdog | 16mV

#### Charge Voltage Limit
> Offset: 3.840V
> Range: 3.840V - 4.608V (`110000`)
> Default: 4.208V (`010111`)
> Note:
> VREG > `110000` (4.608V) is clamped to register value `110000` (4.608V)

### Interpretation

The `bq25896` defaults to a charge voltage limit of 4.208V (`010111`) as noted above.  The charge voltage limit gets set by taking the binary components and adding them up.
[... 76 lines cut ...]

### Datasheet on Charging Voltage

If going for option #**1** above, adjusting the battery gauge's Charging Voltage, section `2.21 Charging Voltage` on page 24 describes some detail:

> ChargingVoltage(): 0x30 and 0x31
>
>  This read-only function returns an unsigned integer value of the desired charging voltage of the battery. A
value of 65,535 indicates that the battery is requesting the maximum voltage from the battery charger.

As this value is read-only, the configuration must be specified as part of the Flipper's `bq27220_init()` function within `Control_ENTER_CFG_UPDATE`/`Control_EXIT_CFG_UPDATE_REINIT`, involving resetting the battery gauge chip whenever the charging voltage limit is adjusted  [... cut ...]
Class | Subclass | Address | Name | Type | Min Value | Default | Units
------|----------|---------|------|------|-----------|---------|------
Configuration | Charge | 0x91FD | Charging Voltage | I2 | 0 | 4600 | 4200 | mV

Of note, **this is separate from the gauge configuration** and **requires additional initialization code.**

### How to reconfigure charging voltage

…as this was already getting quite long, I stopped looking at this point.  I'm happy to continue if this is the preferred method.

</details>

skotopes, 2022-07-06:
Couple things:

- battery is guaranteed to last more than 6 years, in real life it should be closer to 10+
- our battery vendor never did undercharge tests, we can not tell how long battery life will be if undercharged
- biggest impact on battery health is keeping it discharged, which is much harder to control
- if I'd implement it I'll change charger configuration, that gives more predictable results
- gauge data is not very reliable, in some cases it may give offseted values

In my opinion battery life prolongation in that way is not worth it.

digitalcircuit, 2022-07-07:
Firstly, thank you for taking the time to read and respond!  I appreciate all of you are very busy and the work y'all have done is awesome!

I've experimented further, successfully tested changing the charger configuration, and tried to better explain myself.  I expected to be doing the work to implement this, I just want to make sure I'm pursuing the right path.

## Usefulness of charge limiting
### Rationale

I agree with you for folks who are carrying their Flipper with them everywhere (akin to a mobile phone), and/or not paying as close attention to the battery level.  The bigger concern for them is discharging the battery too far.

**The current behavior is a good default!**

In my case, though, I don't travel much, so I'd find it easier to leave the Flipper connected to USB power (via a charger or my computer for the serial console) and just disconnect when I go somewhere, ensuring I never have to think about keeping the battery from discharging too much.

However, with the current battery charging behavior, keeping the Flipper plugged in would try to continually keep the battery at near full charge - 4108mV to 4208mV (based on `bq25896`'s default strict 100mV recharge threshold with `VRECHG` = 0).  This does not seem to follow the typical usage pattern most cells are optimized for.

6+ years is a long time, but I tend to use electronic devices longer than average (e.g. I daily-drive a 2014 laptop where the battery has degraded to "70% health", which actually means "about 45 minutes usage at low brightness with little warning before hard shutdown").  If the Flipper battery was easily replaceable, I wouldn't be as concerned, but from watching a teardown video the battery appears to be buried in the middle of the device, with delicate components, plastic clips, tiny screws, and a bit of adhesive in the way.

### Battery manufacturer/model?

Can you share the battery manufacturer and model so I may look up the cell datasheet?  I couldn't find the details [in the Flipper hardware documentation](https://docs.flipperzero.one/development/hardware/schematic ).

(I recognize you might not be allowed to share the actual datasheet, but I might be able to find it online.)

### Optional context
#### Cell manufacturers

I ask since the datasheets for [some well-regarded high capacity lithium-ion cells (see "18650 The good cells")](https://lygte-info.dk/info/BatteriesRidiculousRatings%20UK.html ) make note of reducing charging voltage for use cases where staying charged is more common:

* [The datasheet for the Samsung 35E](https://cdn.shopify.com/s/files/1/0481/9678/0183/files/samsung_35e_data_sheet.pdf?v=1605015771) (found via [a web store](https://www.18650batterystore.com/collections/samsung-18650-batteries/products/samsung-35e )), suggests on page 14 limiting the charging voltage to 4.00v for when the battery is usually connected to power long term (UPS situation), and 4.10v for electric vehicles (a situation where capacity is critical - the limit might be for battery longevity or for safety reasons).
  * Samsung also suggests on page 11 to store at a lower charge for long-term storage "because storage at higher voltage may cause loss of characteristics."  This would apply to long-term primarily keeping the battery fully charged versus partially charged.  Elsewhere in the datasheet suggests even as low as the ex-factory charge (30%, 3.49v - 3.69v) would maintain capacity.
* The [Sanyo/Panasonic NCR18650GA cell datasheet](https://cdn.shopify.com/s/files/1/0481/9678/0183/files/60f5bf9a9fbf95001ac60005-GA6W.pdf?v=1626883833 ) (also found via [a web store](https://www.18650batterystore.com/products/panasonic-ncr18650ga-ga6 )) recommends lowering the charging voltage to 4.15v from 4.20v if charging voltage isn't adjusted according to State of Health, advising on page 10 under "6. Charging Control for Life End" that "Safety and cycle characteristics of Lithium-ion batteries can be improved by reducing the voltage lower than the rated charging voltage."
  * As on page 16, Sanyo also recommends a partially charged state for long term storage (30% / as shipped from factory).

"As shipped from factory" storage charge is too low for daily use, but even a 3.8v charge - the lowest the `bq25896` will allow - seems to provide over a full day of Flipper usage before it even reaches factory voltage.

Higher-end round cell chargers do have a ["storage" mode that charges to about 3.8v - see "Store" near bottom](https://lygte-info.dk/info/roundCellChargerIndex%20UK.html ).

With this in mind, and [considering Table 4 on the more general Battery University page](https://batteryuniversity.com/article/bu-808-how-to-prolong-lithium-based-batteries ), it seems that limiting the charge to 4.00v or so **if one frequently connects (or stays connected) to a charger** could meaningfully increase the battery longevity.

#### Device manufacturers using cells

Other device manufacturers have implemented methods to try to adapt to folks who tend to leave things plugged in:

Dell implemented an adaptive charging algorithm that tracks whether one tends to keep a Dell laptop plugged in versus unplugging and discharging often, and it will reduce the maximum charge if it's normally plugged in: [Reddit link with linked photos](https://www.reddit.com/r/Dell/comments/sffydv/battery_mode_primarily_ac_use_or_custom/ ), [Dell support forums](https://www.dell.com/community/Latitude/Battery-charging-modes/td-p/6210905 ).

[... cut ...]
The question then becomes if enough folks prefer to keep their Flippers on a charger/controlled via USB for integrating this into the Flipper firmware.

I'm happy to put in the effort to add a setting, UI to control it, and some "charge voltage is limited" graphical interface indicator to minimize support hassle, plus going back-and-forth to ensure the code meets all standards.

However, I'd understand if you/Flipper team are not open to accepting any code that implements this - all code does have a maintenance burden.

digitalcircuit, 2022-08-13:
# Design version 2

I'm taking another look at this since, after recent firmware changes, charge limiting is now the only reason I'm building my own firmware instead of using the official builds.

Following @skotopes 's advice, I've focused on adjusting the `bq25896`'s charging voltage.  As such, it's not possible (nor desired) to match this up to specific battery charge percentages in the UI, so I've reworked my design drafts.

I'm open to feedback or suggestions.

I'll look at incrementally implementing this as a new pull request built on top of [my pull request to also check the charging IC if it says charging is done](https://github.com/flipperdevices/flipperzero-firmware/pull/1378).

## `Desktop`/battery icon when charging is limited
[image]

The charged portion of the battery bar is modified to have a cross-hatch pattern in the middle, signaling that the battery charge voltage has been modified.

This will indicate to anyone trying to provide support that charge limiting is activated (if they know), or something about the battery is modified (if they don't know).

***NOTE:** This was changed from the initial design as it's not possible to reliably determine what percentage of the battery is unavailable when limiting via reducing battery charging voltage.*

## `Power` menu adds configuration
[image]
> `Battery Info`
> `Reboot`
> `Power OFF`
> `Limit charging: < OFF >`

In `Settings` → `Power`, there will be a new menu option titled `Limit charging` with the following settings:
* `OFF`
  * This is the default setting and corresponds to `4.2v`
* `4.1v`
* `4.0v`
* `3.9v`
* `3.8v`

***NOTE:** Alternatively, this could be made into a new scene with explanatory text, graphics, and a warning to not let the battery fully discharge.*

### Handling low battery situations
**If the battery ever reaches a critical point (0% capacity remaining), `Limit charging` will be automatically reset to `OFF`** so the next charge will go up to 4.2 volts as usual.

This helps ensure that if someone enables this feature then neglects to keep their Flipper charged, it won't risk damaging the battery further.

If someone enables this feature, it is expected that they will be charging their Flipper frequently.

## `Battery Info` reminds of charge limit when charged
[image]
> `Charged!`
> `Limited to 3.8v`

If charge limiting is enabled, the `Battery Info` screen will add an additional line saying `Limited to [charge voltage]` when charging is complete to remind the user that they enabled charge limiting.

This helps avoid confusion when the battery shows as charged despite not being 100% full.

digitalcircuit, 2022-08-16:
## Work-in-progress firmware build

I have implemented all the non-configuration graphical changes in [my `ft-batt-charge-config` branch on my `flipperzero-firmware` fork](https://github.com/flipperdevices/flipperzero-firmware/compare/dev...digitalcircuit:flipperzero-firmware:ft-batt-charge-config).

As the settings aren't yet implemented, it is currently hard-coded to `3.888v` for testing purposes.

I'll try to look into creating a power settings configuration soon (it's more complicated than I initially expected).  I'll also want to add the automatic disabling of charge limiting if the battery level ever drops to critical.

Here's some live screenshots:

### `Desktop`/battery icon when charging is limited
[image]

The cross-hatch pattern only applies when the battery charging voltage is modified.

### `Battery Info` reminds of charge limit when charged
[image]
> `Charged!`
> `Limited`
> `to 3.8v`

If charge limiting is enabled, the `Battery Info` screen adds an additional line saying `Limited to [charge voltage]` when charging is complete to remind the user that they enabled charge limiting.

digitalcircuit, 2022-08-19:
# Design revision 2.1

After further thought, this is a confusing setting to expose as just `Limit charging: < voltage >`.  I've drafted up a new configuration screen that better explains what this setting does and how to set it.

**NOTE₁:** I have not yet implemented this; the [`ft-batt-charge-config` branch on my fork](https://github.com/flipperdevices/flipperzero-firmware/compare/dev...digitalcircuit:flipperzero-firmware:ft-batt-charge-config) still only implements [the non-configuration UI](https://github.com/flipperdevices/flipperzero-firmware/issues/1158#issuecomment-1216118510 ).

Modifying the Power application to implement settings involves significant changes to the application (*e.g. using a message queue, mimicking the Notification app which manages LED/LCD brightness*), so I want to check if I'm going the right direction first.

**I welcome any feedback on the preferred way to modify the Power app to support loading/saving `/int/.power.settings`**, especially if it differs from copying the Notification app implementation.

## Power menu adds configuration, revised
[image]
> **Limit battery charging:**
> \[*slider with 4 selectable values*\] `3.9V` `4.0V` `4.1V` **`4.2V`**
> \[*icon of an AC power cord*\] « — » \[*icon of a battery*\]
> Always on charger — Full charge, portable use

**NOTE₂:** The `Limit battery charging:` text will be **bold**, like the text for `Primary favorite app:` in the Desktop settings UI.  I don't currently have a good computer approximation of the bold pixel font used by Flipper.

The `Left`/`Right` arrows will change the charging voltage.  `Back` will exit this menu (saving changes).

The AC power cord icon will be added as a new asset.  The battery icon is copied from the Battery Info screen.  This scene/view will involve custom canvas drawing calls, similar to the `Battery Info` screen.

I removed the `3.8V` setting, so the lowest configurable charge level of `3.9V` [ensures that half (40-60%) of the lithium-ion battery capacity will be available at minimum while still providing an estimated 6-8× increase in battery charge cycles](https://batteryuniversity.com/learn/article/how_to_prolong_lithium_based_batteries ).

digitalcircuit, 2022-08-25:
### Brief update for those following along

Another contributor [has already refactored the Power application to support saving/loading settings](https://github.com/flipperdevices/flipperzero-firmware/pull/1647 ).  If merged, their work would simplify my task as I'd just need to focus on the new UI and the new setting.

However, if [the Flipper team decides that the minimum 4Kb internal storage cost](https://github.com/flipperdevices/flipperzero-firmware/pull/1647#pullrequestreview-1082875463 ) to add two settings (that pull request *plus* my future pull request) is too expensive for the Flipper's limited internal storage, then this feature might not be practical for me to add.  Without settings, it would only be a temporary change (being reset to 4.2v *on every reboot*).

I'll continue to plan and implement, but I might wait for the Flipper developer team to decide on whether they'll allow the storage cost of adding the `/int/.power.settings` file before I get too far along.

D0han, 2022-09-17:
Awesome work!
Can you also add easy to trigger one-off option to charge Flipper fully?
The use case would be pretty much same as yours, but also to easily prepare for a field trip with more battery time.

digitalcircuit, 2022-09-19:
@D0han I think a one-off option to fully charge the Flipper while keeping charge limiting enabled is a good idea.

However, to help keep this task manageable, I don't think I'll be implementing that in my initial pull request.

When I've implemented the current design, it will be possible to manually set the charge limit to 4.2v to fully charge, then when you head off on the trip, set the limit back down when unplugging - no need to wait for the Flipper to discharge.  Hopefully, the lack of the crosshatch pattern on the desktop battery capacity bar will help with remembering that the charge limit is not enabled.

digitalcircuit, 2022-11-30:
For anyone following along, I've gotten everything except for the actual configuration UI implemented in [PR #2063, "power: Add API to get/set battery charging voltage"](https://github.com/flipperdevices/flipperzero-firmware/pull/2063).

If you build that pull request, you can then add and build your own custom tiny Flipper application or service (as described in the verification steps) to set the battery charging voltage limit.  I'll add the integrated settings UI once [PR #1647, "Automatic shutdown on idle"](https://github.com/flipperdevices/flipperzero-firmware/pull/1647) is merged, providing the settings loading/saving backend.

If you have the know-how to build the firmware yourself, help with testing would be appreciated.

digitalcircuit, 2024-06-14:
@skorokithakis Since you asked, here's the details on charge voltage limiting:

Right now, it requires building either a service integrated to the system (my temporary approach), which will work on reboot but means maintaining your own builds, or you can write a standalone third-party Flipper App to call the API for you (not tested).

My changes for system-level integration are here, setting a quite reduced 3.9v (always plugged in):
https://github.com/flipperdevices/flipperzero-firmware/compare/dev...digitalcircuit:flipperzero-firmware:ft-batt-charge-config

There's some old screenshots in the initial API PR:
[FuriHal, Power, UnitTests: battery charging voltage limit API (#2063)](https://github.com/flipperdevices/flipperzero-firmware/pull/2063)

skorokithakis, 2024-06-14:
Ah, I was hoping there would be a console command, but I will try to write an app, thanks!

AkechiShiro, 2025-04-25:
Hey @skorokithakis have you been able to write an app for this if so could you share a link to it ?

skorokithakis, 2025-04-25:
No, I haven't gotten around to anything, sorry 😕

D0han, 2025-04-26:
Momentum-Firmware has implemented charge-cap

digitalcircuit, 2025-04-27:
The charge limit API is already implemented (`furi_hal_power_set_battery_charge_voltage_limit()`), we just need a user interface to set it.  I'm guessing that's what Momentum-Firmware has done.  It might be worthwhile asking if the Momentum folks would contribute the setting back upstream?

(I had still been waiting on the [Automatic Shutdown on Idle PR](https://github.com/flipperdevices/flipperzero-firmware/pull/1647) since that implements a settings backend to build on top of.)