flipperdevices/flipperzero-firmware #1160: Bank/EMV card reading never stops and shows no progress, even when card is missing.

grymoire, 2022-04-24:
I was scanning an American Express card.
The flipper  recognized it as an EMV card. So I loaded the app "Run EMV app" and placed the flipper on top of the card. The device showed "Reading bank card". I left it there.  The display and LED continued to run. (I had the flipper plugged into my laptop to make sure there was enough power.).

I returned 15 minutes later - and there was no change.

I removed the card and the flipper continued "reading" the card. There was no indication that it realized the EMV card was no longer there.

I think it should either show that progress is being made (such as cracking a MiFare DES card keys), or it should realize the card is missing. It looks like it's trying to find encrypted data, but if there is no data as all - is should indicate that the card is no longer there.

Perhaps there needs to be a clear distinction (in the GUI?) that there is a difference between detecting the existence of a card and decoding it? I realize that some cards are trivial to detect/read, and some require specialized apps. Perhaps a small icon that flashes when a card is present or not?

skotopes, 2022-04-25:
Hi,
We need that type of card tell exactly what's wrong.

Couple more things that can help to fix this issue:
- Used firmware version
- Logs from device. (Either from UART on gpio, or `log` command in cli)

grymoire, 2022-04-25:
0.56.1 release

Logs with comments

>: log
Press CTRL+C to stop...
3026383 [D][DolphinState]: icounter 117, butthurt 0
3030993 [D][DolphinState]: icounter 117, butthurt 0

[NFC-A Card detected  - press more]
[run compatible app]

log

3026383 [D][DolphinState]: icounter 117, butthurt 0
3030993 [D][DolphinState]: icounter 117, butthurt 0
3060994 [I][Dolphin]: Flush stats
3060995 [I][SavedStruct]: Saving "/int/dolphin.state"
3060998 [D][StorageInt]: Device erase: page 36, translated page: ca
3061003 [D][StorageInt]: Device sync: skipping, cause
3061004 [I][DolphinState]: State saved

[Found emv card]

3221355 [D][NfcWorker]: Can't find any cards
3222379 [D][NfcWorker]: Can't find any cards
3222571 [D][Emv]: Send select PPSE
3222631 [D][DolphinState]: icounter 117, butthurt 0

[run app]
[and it repeats this message]

3325201 [D][Emv]: Start application
3325230 [D][Emv]: Get proccessing options
3325470 [D][Emv]: Send select PPSE
3325513 [D][Emv]: Start application
3325542 [D][Emv]: Get proccessing options
3325784 [D][Emv]: Send select PPSE
3325828 [D][Emv]: Start application
3325857 [D][Emv]: Get proccessing options
3326096 [D][Em[Emv]: Send select PPSE
3325201 [D][Emv]: Start application
3325230 [D][Emv]: Get proccessing options
3325470 [D][Emv]: Send select PPSE
3325513 [D][Emv]: Start application
3325542 [D][Emv]: Get proccessing options
3325784 [D][Emv]: Send select PPSE
3325828 [D][Emv]: Start application
3325857 [D][Emv]: Get proccessing options
3326096 [D][Emv]: Send select PPSE
3326138 [D][Emv]: Start applicationv]: Send select PPSE
3326138 [D][Emv]: Start application
[I let it run for 15 minutes uninterrupted. The display never changed]

3386269 [D][Emv]: Start application
3386303 [D][Emv]: Get proccessing options
3386563 [D][Emv]: Send select PPSE
3386829 [E][Emv]: Failed to parse application
3387028 [D][Emv]: Send select PPSE
3387054 [D][Emv]: Start application
3387072 [D][Emv]: Get proccessing options
3387291 [D][Emv]: Send select PPSE

And finally I removed the card

3390471 [E][Emv]: Failed to read PAN or PDOL
3390673 [D][Emv]: Send select PPSE
3390719 [D][Emv]: Start application
3390746 [D][Emv]: Get proccessing options
3390980 [D][Emv]: Send select PPSE
3391036 [D][Emv]: Start application
3391087 [D][Emv]: Get proccessing options
3392416 [D][NfcWorker]: Can't find any cards
3393440 [D][NfcWorker]: Can't find any cards
3394464 [D][NfcWorker]: Can't find any cards
3395488 [D][NfcWorker]: Can't find any cards

and the display still says Reading Bank Card

gornekich, 2022-04-29:
Hi!
The reason behind endless bank card reading is that flipper can't parse PAN number from responses it gets. Recently I added trace logs for EMV reader, so you can set debug level Trace and see the raw data Flipper gets. After that you can provide logs here or use TLV parser (for example https://emvlab.org/tlvutils/) to see EMV TAGs thar flipper doesn't support yet.

makeworld-the-better-one, 2022-05-06:
What line in the trace output will provide the PAN number so that I can paste it here? I don't want to paste the full log as that will contain private information.

grymoire, 2022-05-10:
My main concern was in the user interface.  There was no indication that a card is present or missing. That is very important if reading the card will take more than a few seconds. Second - there is no indication of completeness of the reading, such as the one used for cards with DES keys. If the flipper can't ever decode a card -  indicate this. If it needs more time - also indicate this.  And perhaps it's unsure - this could also be useful indication. But showing the same screen for 15 minutes gives the user no indication what they should do next - try another card, wait, give up, move the card closer, etc.

SnowLeopard71, 2022-05-23:
@grymoire  There are updates to the EMV code in the _development_ release channel (in _qflipper_) if you would like to try them.  Or you can wait until the changes make it to _release-candidate_.

skotopes, 2022-07-10:
Please retest with latest dev and reopen if issue is still there.