flipperdevices/flipperzero-firmware #1162: Folders for badusb scripts

grymoire, 2022-04-24:
I wanted to install several dozen badusb (rubberducky) scripts, that I wanted grouped by category.  I placed them on the SD card, in different folders. 

However, when I looked for these scripts, they were not there and the folder did not show up. I had expected to be able to browse into each folder to find the appropriate script.

joshualibrarian, 2022-04-24:
After seeing this issue as well, someone in the discord pointed out to me that while you're right that from within the badusb app (as well as the others) you can't see the files, but if you use the "browser" (clicking down from the main screen and the right/left to switch between files for different applications).  You'll see them all there and you can click on them to run them.

skotopes, 2022-04-25:
Hi, yes we are aware if this issue, we planning to rework file select dialog to be able to open subfolder.

BestUsername, 2022-05-30:
In addition to this, I've found that you cannot open files in subfolders from the browser, either.
The issue seems to be that the function "bad_usb_app_alloc" uses "path_extract_filename_no_ext" for the filename, but that function removes any path information. So Flipper tries to open the filename in the current working directory, which obviously fails.
If the path (either absolute or relative) is included, it might be a good idea to bump up the value of BAD_USB_FILE_NAME_LEN (currently at 40). Or maybe the path should be stored in an additional field?

skotopes, 2022-06-01:
Nice )

closed 2022-06-01 by skotopes