flipperdevices/flipperzero-firmware #1169: Saved MiFare Ultralight NFC not working to target read device

idalisdesalvo, 2022-04-26:
**Describe the bug**
Saved MiFare Ultralight NFC (Amiibo) not working to target read device (Nintendo Switch)

**To Reproduce**
Steps to reproduce the behavior:
1. Target read device (Nintendo Switch) is put into read mode (Read Amiibo)
2. Flipper is navigated to NFC->Saved->[file].nfc->Emulate Ultralight
3. Target device (Nintendo Switch) reports "Failed to read Amiibo".
4. Flashing firmware to 55.1 (from 56.1) results in working as expected.

**Expected behavior**
Target read device reads NFC tag (Amiibo) from Flipper saved NFC file.

**Logs**
Add debug logs

**Target**
Target read device is Nintendo Switch HAC-001(-01)

**Additional context**
Add any other context about the problem here.

jlaughter, 2022-04-27:
Confirmed above behavior in 56.1, 56.1RC and Dev19f42c53.  Rolling back to 55.1 restores emulate functionality.

gornekich, 2022-04-28:
Fix in https://github.com/flipperdevices/flipperzero-firmware/pull/1179
Soon will be in release

skotopes, 2022-04-28:
fixed in dev, will be include in release 0.57