flipperdevices/flipperzero-firmware #1171: Program multi-key IR inputs

edbrannin, 2022-04-26:
**Is your feature request related to a problem? Please describe.**

I'd like to save a sequence of IR signals to replay from the Flipper, like:

- TV Input, TV 2
- TV power on, Receiver power on, Receiver input X

**Describe the solution you'd like**

Today: When recording a new IR payload, as soon as the Flipper recognizes a signal, it stops listening and prompts to [RETRY, SEND, SAVE].

Maybe add an ADD option to make the Flipper capture a second signal to play after the first?

**Describe alternatives you've considered**

Other options include:

- Crafting macros from the IR app (more flexibility, but there are UX challenges; we can't even reorder custom remote buttons yet)
- Crafting macros from the Mobile apps (Simpler UX, but requires an additional device for setup)
- Sending IR-app-recorded signals from a compiled Application (those can't be sideloaded yet, and it would limit this sort of thing to programmers)

**Additional context**

[`u/skotozavr`](https://www.reddit.com/user/skotozavr/) [suggested I report this issue from Reddit](https://www.reddit.com/r/flipperzero/comments/tzy80f/can_you_do_multiplea_series_of_infrared_replays/)

skotopes, 2022-04-28:
Crafting is possible. For example raw payloads can be gathered together. We'll think about it.

DrEverr, 2022-08-20:
I also thought about this feature. It can be useful for saving favourite programs, where there is not enough time frame on TV to scroll all buttons on flipper to input desire program.

gelin, 2022-09-08:
I have another view of this issue, from the `.ir` files perspective :)

Currently the `tv.ir` file from `/infrared/assets` has multiple definitions of the same button.
For example:
```
Filetype: IR library file
Version: 1
# 
name: POWER
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 15 00 00 00
# 
name: POWER
type: parsed
protocol: SIRC
address: 10 00 00 00
command: 15 00 00 00
```

However, when I try to do the same for my recorded IR signals, for example:
```
Filetype: IR signals file
Version: 1
# 
name: Cool 25
type: raw
frequency: 38000
duty_cycle: 0.330000
data: /*skipped*/
# 
name: Cool 25
type: raw
frequency: 38000
duty_cycle: 0.330000
data: /*skipped*/
# 
```
The Flipper's UI displays two buttons with the same name "Cool 25".

My intent is to craft an "universal" AC control. So, I recorded the signals to set the "cool to 25°" mode from multiple ACs. And now I want to send multiple IR sequences when I press the single button.

So, a simple improvement can be: make the behavior of "IR signals file" similar to "IR library file", if multiple buttons with the same name are defined, display only one button, but send multiple IR signals when the button is pressed.

robertcedwards, 2022-12-23:
Same is happening to me, trying to use multiple definitions of one button with the same name, and it's not replaying these like the universal remote. Also tried to delete the universal remote and use my button definitions to playback, but to no avail.

HacktheTime, 2024-01-24:
Also problematic when you have something like an Epson EB 725W which wants a reset signal first.