flipperdevices/flipperzero-firmware #1172: Stopwatch and Clock

RogueMaster, 2022-04-27:
Would be nice to have a stopwatch and a clock. It could even be an accessible feature in the unimplemented dumb mode.

skotopes, 2022-04-28:
Possible, low priority. Anyone who will want to implement are welcomed to do that.

zhovner, 2022-05-08:
There is no guarantees that Flipper's internal clock is synced and  accurate. Thats why we not implement clock in it. 

About stopwatch -- doesn't your phone can do this better?

Suggested to close.

RogueMaster, 2022-05-08:
I have a clock and stop watch application now. Seems to work well. Not sure why you would suggest to close such a valuable feature that is so easy to add. Clock is now my main favorite/first app on my flipper and my most used.

RogueMaster, 2022-05-08:
I will confirm with CompaqDisc on his plans to submit his clock app. I modified his clock app to have the stopwatch.

RogueMaster, 2022-05-08:
Here is the code for clock and stopwatch. https://gist.github.com/RogueMaster/ed2e16647a0ade3372dcbeb078507517 
I can remove sound alerts if it is preferred

skotopes, 2022-05-10:
furi_hal_speaker is meant to be used throw notification service.

RogueMaster, 2022-05-10:
How do you recommend it be changed? If that is what you are suggesting.

I am using it to throw a notification yes. Service was not needed as it is operating as desired without bug/compile errors.

RogueMaster, 2022-05-10:
Updated gist to remove the sound notifications... https://gist.github.com/RogueMaster/ed2e16647a0ade3372dcbeb078507517

RogueMaster, 2022-05-17:
I have updated the sound notifications so it uses the the proper process. Please see code here: https://github.com/RogueMaster/flipperzero-firmware-wPlugins/blob/dev/applications/clock_app/clock_app.c

MCWertGaming, 2022-07-28:
Would be cool for Sure! I might look into this. Is there a preferred way to add applications? I know that currently they are compiled into the firmware, but is it possible to create applications outside of the firmware tree? Would be nice for applications that are not accepted from the maintainers, but are still developed and maintained by the community.

RogueMaster, 2022-07-28:
@MCWertGaming check my github 😎 This issue is being "fixed" by #1405

MCWertGaming, 2022-07-28:
Really nice, thanks!

kowalski7cc, 2022-09-08:
Stepping in, I made a quick figma how the feature could work: https://www.figma.com/file/qwhvP7uKV1kKmy08PzGjM7/clock_app?node-id=0%3A1

hedger, 2022-09-21:
Now this can be implemented with freslhy released FAP package support outside our code base.

closed 2022-09-21 by hedger