flipperdevices/flipperzero-firmware #1173: NFC over USB (LibNFC) functionality

tusing, 2022-04-27:
In [this](https://www.reddit.com/r/flipperzero/comments/is31re/mifare_attacks_with_flipper/) Reddit issue, @zhovner mentions the Flipper Zero would be able to be used as a "regular USB NFC adapter along with LibNFC." He also provided this diagram:

[image]

I was wondering if this functionality is implemented yet, and if there are any plans to implement it? I was trying to run some attacks on my Mifare Classic that require a powerful CPU, but I cannot seem to find the NFC over USB option in my own Zero that just arrived.

GMMan, 2022-05-07:
The original idea of passing through commands directly to the NFC hardware sounds like it'd work, and I think is something there are already examples for (with other chips) in libnfc. I'm willing to take a look at this as I have a couple of Mifare Classic cards I want to crack too, though probably not for another few weeks. (I started trying to implement a new driver for a PCSC reader until I found out it doesn't give low level enough control.)

skotopes, 2022-05-16:
> The original idea of passing through commands directly to the NFC hardware sounds like it'd work, and I think is something there are already examples for (with other chips) in libnfc. I'm willing to [...]

There are some timing issues that are going to break everything. We've made couple test and it looks hardly feasible.

SnowLeopard71, 2022-05-22:
@skotopes 
Maybe the [*September progress* blog post](https://blog.flipperzero.one/september-progress/) could be updated with a pointer to this issue?   It's where the image is from.
If I hadn't found this closed issue accidentally, I would have been waiting for the *NFC over USB* functionality to be implemented.  Thanks.

jevinskie, 2022-06-04:
I’ve successfully implemented NFC communications (not using libnfc though) from a host computer through USB to a MCU connected to the ST25R3916 via SPI. I was able to simple pass through most commands over a USB-SPI bridge while handling a few timing sensitive operations in the MCU firmware and custom USB commands.

GMMan, 2022-06-15:
> I’ve successfully implemented NFC communications (not using libnfc though) from a host computer through USB to a MCU connected to the ST25R3916 via SPI. I was able to simple pass through most [...]

Any plans on sharing this work?

jevinskie, 2022-06-15:
> Any plans on sharing this work?

Sorry I can’t, it was done for work.

eqvinox, 2022-08-30:
> > Any plans on sharing this work?
> 
> Sorry I can’t, it was done for work.

@jevinskie do you feel you can share the list of timing sensitive operations you had to implement directly in the MCU? That might make it easier for someone with limited NFC knowledge to implement this.

tusing, 2022-08-31:
@skotopes If someone has successfully managed to implement this, even if they can't share it due to NDA, should we re-open this issue?

skotopes, 2022-08-31:
@tusing just try to read your comment one more time and answer this question by your self. 
PS: Also @jevinskie explicitly mentioned that libnfc is not used.

LevitatingBusinessMan, 2023-04-02:
If the cli for subghz were more flexible, would that not be enough to do some more advanced Mifare attacks?

skotopes, 2023-04-02:
It depends.