flipperdevices/flipperzero-firmware #1176: Compiled Icons are not produced as part of build process in docker

leeroywking, 2022-04-27:
**Describe the bug**
I am attempting to create a new "Application" for the D.0 and after making the pixel art for the main menu icon I was getting the following error after trying to wire everything up. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create new Application
2. Add Icons to ./assets/icons/<desired location>
3. run `docker-compose up -d`
4. run `docker-compose `docker-compose exec dev make`
5. run terminates with error to this effect (full run log available at above link)

```
/project/applications/applications.c:185:82: error: 'A_TextReader_14' undeclared here (not in a function)
  185 |     {.app = text_reader_app, .name = "Text Reader", .stack_size = 1024, .icon = &A_TextReader_14},
      |                                                                                  ^~~~~~~~~~~~~~~
        CC       applications/gpio/views/gpio_usb_uart.c
        CC       applications/gpio/views/gpio_test.c
make[1]: *** [/project/make/rules.mk:71: .obj/f7-firmware//project/applications/applications.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/project/firmware'
make: *** [Makefile:68: firmware_all] Error 2
```

**Expected behavior**
I expect icons to be compiled into compiled resources as part of a normal build if not already present.

**Logs**
https://github.com/leeroywking/flipperzero-firmware/blob/icon-build-error/steps-To-Reproduce%2Blog.txt

**Additional context**
I was able to fix this error for my own use by entering the container and running the asset.py script manually with the following command.
`python3 ./scripts/assets.py icons ./assets/icons/ ./assets/compiled/`

leeroywking, 2022-04-27:
I'd be happy to submit a proposed fix to this I am extremely new to make and C lang stuff and I'm just coming online to the project from a dev standpoint but if someone can describe the desired fix to this I'm happy to hack through it and submit it for PR I just wanna make sure its a desired fix

skotopes, 2022-04-28:
Try following:

```
make -C assets clean 
make -C assets all
```

Also please read ReadMe.md in assets folder, it should explain process.

closed 2022-04-28 by skotopes

leeroywking, 2022-04-28:
Oh that's actually not how the readme describes that and when I did a search on the code base I didn't find that anywhere in there repository. Here is a pull request to update the relevant readme so that it is documented how you describe
https://github.com/flipperdevices/flipperzero-firmware/pull/1181