flipperdevices/flipperzero-firmware #1177: Amiibo no longer works

DogmaBomb, 2022-04-28:
**Describe the bug**
Before updating to 0.56.1 the amiibo were loading just fine to my switch, 
now with the new update to 0.56.1 the Amiibos will not load in after 
multiple tries.

**To Reproduce**
Steps to reproduce the behavior:
Scanned NFC with Zelda BOTW
Worked before
Doesn't work with firmware 0.56.1

**Expected behavior**
Changes made to official firmware

**Logs**
None

**Target**
Nintendo Switch

**Additional context**
Will attempt to run previous firmware to ensure amiibos still work

gornekich, 2022-04-28:
Fix in https://github.com/flipperdevices/flipperzero-firmware/pull/1179
Soon will be in release

skotopes, 2022-04-28:
Fixed in dev, will be in release 0.57

closed 2022-04-28 by skotopes